NanoVNA-Saver/nanovna-saver #142: Calibration averaging

TheOriginalFreddoFrog, 2020-01-25:
I work here in a RF noisy location and have noted that sometimes I see unusual glitches that are out of the ball park when I read from my nanovna-h firmware based device. I perform another calibration and these unusual glitches are gone.

Would it be possible and perhaps desirable to implement an averaging facility on calibration? 

i.e.1: run multiple calibration passes (user configurable) and accept the average?

i.e.2: perhaps even more advanced facility where multiple passes (say 5 minimum) are run and values exceeding 1 standard deviation are eliminated from the averaging process?

mihtjel, 2020-01-25:
Hi,
Averaging, including discarding extreme values, is already a part of nanovna-saver, and can be enabled for in-application calibration. :-)

I (as an application developer) have no way to do this for the on-device calibration. I think memory constraints are a significant issue in trying to implement this in the firmware. But do try asking the firmware developers :-)

TheOriginalFreddoFrog, 2020-01-25:
Mihtjel thanks,

Yes the limitations that some developers have encountered (i.,e, the "Q" branch that does support an on-firmware averaging function) are a direct result of the STM32 MCU's used.

I am not talking about manipulating firmware functions nor dealing with hardware functions except to read raw data off it; these comments/suggestions relate to reading data from the device back into into 3rd-party software - NanoVNA Saver - so "more" can be done and with greater accuracy :-) 

I tune cavities and duplexers.

What I am suggesting is a function for NanoVNA Saver that allows a configurable number of independent calibration passes (i.e. a minimum of say 5 passes) at the calibration phase ... 

Such a method would create a more realistic picture of the environ around the user of the NanoVNA with NanoVNA Saver and this give better results.

....

I do have a question from your comment above:

> is already a part of nanovna-saver, and can be enabled for in-application calibration. 

How can this be enabled and perhaps a little more detail may be required?

I am examining the UI and cannot find the facility. I have not gone near the code yet due to time constraints ... but a little guidance here with that comment would be highly appreciated :-)

mihtjel, 2020-02-02:
If you go to sweep settings, you can set up NanoVNA-Saver to use averaging, and to discard the extreme values. This is also applied to in-application calibration using the calibration assistant.

zarath, 2020-06-27:
Averaging is a basic feature in sweep settings

closed 2020-06-27 by zarath