NanoVNA-Saver/nanovna-saver #147: Case-insensitive entry of frequency units

zstadler, 2020-02-02:
**Is your feature request related to a problem? Please describe.**
While most of the frequency units are case insensitive, `mhz`, `ghz` are not accepted.

**Describe the solution you'd like**
Make the entry of frequency units case-insensitive.

**Describe alternatives you've considered**
In addition, it may be appropriate to reduce the set of accepted SI units, as recommended here:
https://github.com/mihtjel/nanovna-saver/blob/00822f4a9ff7aee4625ffec7751be3ef82576321/test/test_parseFrequency.py#L85-L90

**Additional context**
Currently, both the SI standard `k` and non-standard `K` are allowed for Kilo Hertz. The `Hz` suffix is both case-insensitive and optional. Only `m` and `g` are not acceptable alternatives to `M` and `G`, respectively.

Although `mHz` is a valid SI unit of mili-Hertz, such units of 0.001Hz are irrelevant for NanoVNA...

I will be glad to provide a PR.

zarath, 2020-07-17:
Will be in #4e4905a77a6fff7bc31130b5c46da4fb62fa6ea9

closed 2020-11-01 by zarath

zstadler, 2020-11-02:
Thank you @zarath!