NanoVNA-Saver/nanovna-saver #150: Impedance S11 /Z/ on markers

UlrichD61, 2020-02-05:
Hello,
I'm using Saver primarily to measure impedances on components.
Quite simple to measure and display Real/Imag, and it's great that features be added day by day !!
That's the reason I'm writing:   It would be ideal to see the value of  S11 /Z/ in the markers as well.

In the charts  S11 /Z/ can be selected to display the **graph** of S11 /Z/.
It would be great to also include S11 /Z/ to the selectable entries in "Marker Settings" for showing the S11 /Z/ as a Ohm-number for each markers frequency.

Would be of great help and avoids using Pytharoras for each and every marker.
Thanks !

[image]

yo2naa, 2020-02-06:
It would be a nice addition, please add |Z| to the markers. I've recently measured some CMM chokes and use markers for each band of interest, it would be great to directly read the |Z| on each marker.
Thank you.

zarath, 2020-02-06:
I'll take a look

zarath, 2020-05-16:
Merged in development branch

closed 2020-05-16 by zarath