NanoVNA-Saver/nanovna-saver #151: Save / load markers

yo2naa, 2020-02-06:
**Is your feature request related to a problem? Please describe.**
Each time I open NanoVNA-saver, I need to redefine the markers. I use a lot of markers, for example, one for each of the amateur radio HF bands, some times more than one marker/band.

**Describe the solution you'd like**
After defining the markers (frequencies) to be able to save them and load next time. Multiple sets of markers would be great, saved with different names (example HF markers)

**Describe alternatives you've considered**
To define the markers each time I open the software.

Thank you!

zarath, 2020-02-06:
would like it, too.
but hopefully i get the marker specific code separated, as adding more complexity in main code file wouldn't improve readability and stability in the long term.

zarath, 2020-02-06:
> but hopefully i get the marker specific code separated, as adding more complexity in main code file wouldn't improve readability and stability in the long term.

Which is done now, but please allow some time to implement persistent marker settings...

yo2naa, 2020-02-07:
Many thanks, really appreciate your work!

yo2naa, 2020-10-21:
Kind reminder, TNX

akinad3, 2020-12-28:
this would be really awesome! Running into the same issues as @yo2naa

yo2naa, 2020-12-28:
There will be a problem when changing the number of segments or scan range. In this case, the marker frequencies could be replaced by the closest frequency, and markers outside the scan range should be deleted. 
I would be nice to save a few custom profiles containing a scan range, the number of segments and the associated markers.
Thank you.