NanoVNA-Saver/nanovna-saver #155: Multiple bugs related to array/list handling

gabriel-tenma-white, 2020-02-08:
In SweepWorker::updateData() we have:
```python
data11, data21 = self.applyCalibration([raw_data11], [raw_data21])

self.data11[offset * segment_size + i] = data11
self.data21[offset * segment_size + i] = data21
self.rawData11[offset * segment_size + i] = raw_data11
self.rawData21[offset * segment_size + i] = raw_data21
```
Note that applyCalibration() returns List[Datapoint], so the 2 assignments that follow will put a list into a list, which would later cause a crash in the main application, but another bug in SweepWorker::saveData() causes self.rawData11 to point to the same list as self.data21, so normally the next 2 assignments to self.rawData* overwrites the entry in self.data* again with the raw data!
```python
def saveData(self, frequencies, values11, values21):
   ...
   self.data11, self.data21 = self.applyCalibration(raw_data11, raw_data21)
   self.rawData11 = raw_data11
   self.rawData21 = raw_data21
```
Note that applyCalibration() will return the original list passed to it in some cases, which causes rawData* and data* to be the same arrays.
#115 was another bug that is caused by taking a reference to a list, and the list later being changed. I recall seeing a few more places that had similar issues, but can't remember now.

I think there needs to be some mechanism to prevent this kind of bug popping up again, such as the device and calibration routines returning immutable lists so that stored arrays are guaranteed not to change when you don't expect it to, and you can't reuse the array returned by the HW/calibration library to store later sweeps. Or some way to pass lists "by value" by default like in C++...

mihtjel, 2020-02-08:
Thanks for this. I have noticed some problems in this area of code before, and I think I need to go through it again. :-)

zarath, 2020-06-27:
Stil valid in v0.3.0 and up?

zarath, 2020-07-15:
refactored in v0.3.0 and up

closed 2020-07-15 by zarath