NanoVNA-Saver/nanovna-saver #165: Please add R+jX and |Z| graphs calculated from S21 with G3TXQ method

yo2naa, 2020-02-29:
**Is your feature request related to a problem? Please describe.**
Measuring high impedance with NanoVNA is not reliable. 

**Describe the solution you'd like**
A better method of measuring high impedance to insert the device to be measured between CH0 and CH1, and calculate the impedance using S21 magnitude and phase. This method is described by G3TXQ here http://www.karinya.net/g3txq/chokes/#measurement
The explanation and formulas are here http://www.karinya.net/g3txq/chokes/s21.pdf
[image]

I've measured a 9.8 k small THD resistor using S11 and also the G3TXQ method, the results are much better using G3TXQ method:
[image]

**Describe alternatives you've considered**
Manually copy S21 data into excel for various frequencies and do the graph in excel

Thank you.

satfan52, 2020-03-01:
Good idea! What about measuring very low impedances with accurracy? Can the same technique be used with as much accurracy ?

cozycactus, 2020-03-08:
Good suggestion!

yo2naa, 2020-10-21:
Kind reminder, thanks!

akinad3, 2020-12-28:
this would be great to measure common-mode chokes!

rjordans, 2021-02-18:
To measure very low impedances you'd probably want to have a shunt-through measurement.  You want to have the DUT cause a significant variation in the measured S21 with variation of the component values.  W0QE has a nice video about the different approaches and how they impact the accuracy: https://www.youtube.com/watch?v=1UbEz73FGCU

rjordans, 2021-02-18:
I had some time to play with this and implemented both the R+jX and |Z| graphs.  Some quick tests with a few SMD resistors, capacitors, and a ferrite bead I had on my desk seem to confirm the correct operation.

no3m, 2021-02-18:
> I had some time to play with this and implemented both the R+jX and |Z| graphs. Some quick tests with a few SMD resistors, capacitors, and a ferrite bead I had on my desk seem to confirm the correct operation.

Getting the following errors when sweep segments are > 1:

S21 R+jX series plot enabled:

Traceback (most recent call last):
  File "/home/eric/Public/Downloads/amateur/software/nanovna-saver-rjordans/nanovna-saver/NanoVNASaver/Charts/Frequency.py", line 416, in paintEvent
    self.drawValues(qp)
  File "/home/eric/Public/Downloads/amateur/software/nanovna-saver-rjordans/nanovna-saver/NanoVNASaver/Charts/RI.py", line 206, in drawValues
    max_real = max(8, math.ceil(max_real))
OverflowError: cannot convert float infinity to integer
Aborted (core dumped)

S21 |Z| series plot enabled:

Traceback (most recent call last):
  File "/home/eric/Public/Downloads/amateur/software/nanovna-saver-rjordans/nanovna-saver/NanoVNASaver/Charts/Frequency.py", line 416, in paintEvent
    self.drawValues(qp)
  File "/home/eric/Public/Downloads/amateur/software/nanovna-saver-rjordans/nanovna-saver/NanoVNASaver/Charts/MagnitudeZ.py", line 107, in drawValues
    maxValue = 10*math.ceil(maxValue/10)
OverflowError: cannot convert float infinity to integer
Aborted (core dumped)

rjordans, 2021-02-19:
Hmm, I did test with multiple segments, that does work for me.  But looking at the error I think this may be something different.

My guess is that your calibration is better than mine and you have nothing connected between the two ports.  That would result in S21 going to 0 and the series calculation divides by S21.  This gives an infinite impedance which apparently is causing trouble in the plotting code when it tries to get the automatic vertical scale correctly.

I'll have a look and see how to correct for this, maybe some large fixed upper bound no reasonable impedance values will work though that feels a bit hacky.

no3m, 2021-02-19:
The DUT during these failures was a Murata 5000 series (4.7mH) surface mount CM choke.  Respective ends of the CM choke tied together and in series with Ch0 and Ch1, ground pass-thru.

rjordans, 2021-02-19:
Ok, I've added some code to catch the infinite values and not use them for the plotting.  Hopefully this fixes your issue?  It looks like my initial guess was indeed correct and that the plotting code didn't handle this case properly yet.

no3m, 2021-02-20:
latest changes seem to work fine.

yo2naa, 2021-02-20:
Many thanks for adding this nice feature, really appreciate the effort. 
I'm trying to run the NanoVNASaver.x64 (nanovna-saver.exe) on my Windows 10 Pro 64 but I'm getting some errors:

 File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\NanoVNASaver.py", line 28, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\Windows\__init__.py", line 2, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\Windows\AnalysisWindow.py", line 23, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\Analysis\__init__.py", line 1, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\Analysis\Analysis.py", line 22, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\signal\__init__.py", line 302, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\signal\filter_design.py", line 16, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\optimize\__init__.py", line 421, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\optimize\_shgo.py", line 9, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\spatial\__init__.py", line 107, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.7\x64\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\spatial\transform\__init__.py", line 19, in <module>
  File "rotation.pyx", line 5, in init scipy.spatial.transform.rotation
ModuleNotFoundError: No module named 'scipy.spatial.transform._rotation_groups'

rjordans, 2021-02-21:
@yo2naa That looks like an independent problem from this feature.  Looking at the error messages I believe that there is some component of the scipy library of Python missing in the install.

r2axz, 2021-03-14:
I wrote [this Python script](https://github.com/r2axz/vic) to calculate complex impedances in s11 shunt, s21 series, and s21 shunt through modes. Works OK with NanoVNA as well as with any other VNA that can export Touchstone files. You can probably get the formulas there or use it for testing purposes. I use the script mainly for HF baluns / chokes measurements, but the math to calculate the impedance is universal.

yo2naa, 2021-04-04:
> 
> 
> @yo2naa That looks like an independent problem from this feature. Looking at the error messages I believe that there is some component of the scipy library of Python missing in the install.

Many thanks Roel for all the good work. A friend of mine nanaged to build the exe on my PC, and now I can use the new features. Would you be so kind to add them to the markes too?
[image]
Thank you so much!

akinad3, 2021-04-04:
@rjordans 
Opened a PR for adding the markers, but initial tests with @yo2naa shows differences between marker readings and data on graph (for R+jX series for example):
https://github.com/NanoVNA-Saver/nanovna-saver/pull/384

yo2naa, 2021-04-04:
I can confirm that the graph for S21 R+jX series is unfortunately not OK:
[image]
However, S21 |Z| series graph is perfect:
[image]

The marker values added by akinad3 are OK, thank you:
[image]

rjordans, 2021-04-06:
@akinad3 thanks for adding the markers!

Regarding the weirdness you see, is this also in the graph view or just in the marker values?  And what would be an easy way for me to reproduce that so that I can see if it's an easy fix somewhere?

yo2naa, 2021-04-07:
There is only a minor issue with R scale low limit:
[image]
I think is is caused by negative R from the calculations, due to S21 measurement errors. For example, S21 series R for 69.4 kHz is -0.272 Ohm. Of course this is wrong, but not due to software, but because of wrong S21 measurements. However, the lower limit on the graph is still odd, -3623 Ohm.
[image]

Here is the s2p file:

[7 turn RG142 toroid choke S21.zip](https://github.com/NanoVNA-Saver/nanovna-saver/files/6270264/7.turn.RG142.toroid.choke.S21.zip)

rjordans, 2021-04-08:
Ah, yes, this indeed looks like it comes from inaccuracies in the initial measurement.  I'm unsure though how to handle this nicely but at least the lower bound of the axis could be closer to the measured value.  Looking at your graphs, the plotted label looks a lot like it's related to the data on the other axis, this may simply be a case of the wrong value being printed at that location.

rjordans, 2021-04-23:
Ok, I finally had a bit of time to look into the axis labeling.  It seems that this happens for any real/imaginary plot and was a simple typo to fix!

rjordans, 2021-04-23:
Getting the negative real part printed correctly in the markers also looks like an easy fix. I've added a comment to the pull request on how to get that fixed

yo2naa, 2021-05-01:
Thank you all, working fine now!

PR #376 (2021-02-19): Added measurement graphs showing impedance measured via S21
Fixes #165

- Allows for both |Z| and R + jX display
- Measurements of shunt impedance (accurate for low impedance values)
  as well as series impedance (accurate for high impedance values)
commit c6878fce (2021-02-19): Added measurement graphs showing impedance measured via S21

Fixes #165

- Allows for both |Z| and R + jX display
- Measurements of shunt impedance (accurate for low impedance values)
  as well as series impedance (accurate for high impedance values)
closed 2021-06-20 by zarath