NanoVNA-Saver/nanovna-saver #169: Ver 0.2.2: Continuous sweep broken - Ubuntu 18.04 - sweep freezes after the first occurence

satfan52, 2020-03-15:
Hello,

I have calibrated from 1 to 300 Mhz with 1 segment (3 mhz steps). Single sweep works fine, but when I switch to continuous sweep then the application performs a first sweep ok but then the continuopus sweep freezes, impossible to try to stop it. The application must be killed and restarted. I am using Ubuntu 18.04.

I tried  
python3.7 -m pip uninstall .
python3.7 -m pip install .
but no luck

all info below
Regards
Peter

root@on7yi-ubuntu:/opt/build/nanovna-saver# nanovna
NanoVNASaver 0.2.2
Copyright (C) 2019 Rune B. Broberg
This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/mihtjel/nanovna-saver for further details
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Settings: /home/peter/.config/NanoVNASaver/NanoVNASaver.ini
Traceback (most recent call last):
  File "/opt/build/nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 750, in dataUpdated
    m.findLocation(self.data)
  File "/opt/build/nanovna-saver/NanoVNASaver/Marker.py", line 240, in findLocation
    min_freq = data[0].freq
AttributeError: 'list' object has no attribute 'freq'
Traceback (most recent call last):
  File "/opt/build/nanovna-saver/NanoVNASaver/SweepWorker.py", line 163, in run
    self.updateData(values, values21, i)
  File "/opt/build/nanovna-saver/NanoVNASaver/SweepWorker.py", line 194, in updateData
    freq = self.data11[offset * segment_size + i].freq
AttributeError: 'list' object has no attribute 'freq'

[300mhz.zip](https://github.com/mihtjel/nanovna-saver/files/4334851/300mhz.zip)

[image]

[image]

satfan52, 2020-04-24:
For reasons I don't understand the problem disappeared, I am able  to sweep continuously with 0.22. I am concluding the problem is in my environment so I am closing the ticket and may re-open it  I succeed to understand what are the conditions that make the continuous  sweep "works" or "breaks" in V 0.22

satfan52, 2020-04-24:
Ok in my environment (Ubuntu 18.04) continuous sweep works ONLY in un-calibrated mode. When I load a calibration file previously saved then the Continuous Sweep functionality is broken as explained above in this ticket. I also tried with a FRESH calibration using the "assistant" and the issue is the same, no change. As said before, in version 0.23 I do not have this issue after doing the calibration, but the markers are not exploitable in continuous mode, they get reset to the start frequency at every sweep !!!

satfan52, 2020-05-01:
Today I did a fresh install of the master branch with tag 0.2.2 and the problem dissapeared

satfan52, 2020-05-01:
correction:

Fixed in the windows downloable version 0.22-1
NOT fixed in the master version LINUX (label 0.22)
Partially fixed in the Development version (label 0.23) as markers are not usable in continuous sweep mode

zarath, 2020-06-28:
Outdated

closed 2020-06-28 by zarath