NanoVNA-Saver/nanovna-saver #172: Serial port 'connect/disconnect' button text doesn't change till window event

Mike4U, 2020-04-15:
When the connect button is pressed, it highlights but the disconnect text doesn't show up till some other window is activated.

Steps to reproduce the behavior:
1. On Mac 10.15.4, Python 3.7, NanoVNA Saver 0.2.3, AVNA
2. Click on 'Connect to device'
3. Watch it connect, start sweeping and show plots, even place markers.
4. Button still says 'Connect to device'
5. Click outside the Nano saver window or click the manage button and the connect button will finally change to disconnect.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version: [if applicable]
 - NanoVNA-Saver version: [e.g. 0.1.4]

**Additional context**
Add any other context about the problem here.

zarath, 2020-07-14:
stil valid with v0.3.4 or v0.3.6-rc?

Mike4U, 2020-07-16:
Works as expected with v0.3.4 when connecting to AVNA,  but same issue when disconnecting.  Also same issue when connecting or disconnecting with NanoVNA V2

Haven't tried v0.3.6-rc

Mike4U, 2020-07-30:
finally tried with v0.3.6  binary download for MacOS as I messed up my GitHub somehow.  Crashed when connecting to the AVNA multiple tries.  Functioned with NanoVNA V2 with same issues as original post, maybe worse.  Now the buttons seem to require two clicks to respond.

Mike4U, 2020-08-03:
Tried  0.3.7-rc06 from source.  Button update is still off.  The easiest way to see is with the marker hide button.   Wether this button remains or not it is a good demo of issue that seems common to serial port buttons and drop down.  Sometimes the button displays correctly but gets out of sink after a press or two the show and hide are reversed.  Clicking out of the main window correctly updates the buttons to the correct text.  I'll make separate post for AVNA.

Mike4U, 2020-08-04:
Found a fix.  The issue is with PyQT5 and Mac OS 10.15.x  Here's a link to similar issue [https://stackoverflow.com/questions/56553257/pyinstaller-and-pyqt5-macos-mojave-compatibility-issues](url)
Can't submit but here's the changes.
[image]

Mike4U, 2020-08-04:
and the other:
[image]

Mike4U, 2020-08-04:
Dang the link didn't work;
[https://stackoverflow.com/questions/56553257/pyinstaller-and-pyqt5-macos-mojave-compatibility-issues](url)

Mike4U, 2020-08-04:
Did a pull request ( I think :)

Mike4U, 2020-08-04:
Ya

closed 2020-08-04 by Mike4U