NanoVNA-Saver/nanovna-saver #174: how to make nanovnasaver.exe so small

dorappp, 2020-04-23:
how to make the nanovansaver.exe so samll,I try to use the virtualenv，but it's also too big

zarath, 2020-06-29:
What is small and what is too big? And why this is a Bug?

dorappp, 2020-06-29:
NO,It's not bug,I just want to know how you can install it so small.
when I install some python program include numpy to the .exe, it's always too big. 
Would you like to teache me?

zarath, 2020-06-29:
Is the actual binary (v0.3.2) still smaller then yours?

It's just created with pyinstaller.

dorappp, 2020-06-30:
....OK  Thank you.  I'll try again

closed 2020-06-30 by zarath

dorappp, 2020-07-18:
sorry to bother you.
I'm  confused about the communication protocol between NanoVNASaver and NanoVNA-H.
Waht's the Communication data format of them.Where to get the  relevant information.
Thank you very much