NanoVNA-Saver/nanovna-saver #178: Error in exported S1P file .

Herman-RFguy, 2020-05-10:
Dear all,
I compared 3 S1P files.
All generated with the same way of calibrating, and the same DUT.
one generated with a R&S ZNC-3, one with nano-VNA and nanovna-saver , and one with nanovna and sharp SW.

Calculating from the S1P files back to the marker value readings the R&S ZNC3 and the combination nanoVNA with Sharp succeeds, the combination nanovna with Saver fails.

conclusion: I assume The nanovna-Saver software exports erroneous S1P files.
I enclose all 3 S1P files for investigation.
sharp.s1p.zip
znc3.s1p.zip
nanovna_result.s1p.zip

For a nice application, measuring the complex permeability of toroids by measuring S11
it is important to have good data, so my kind request is a bug fix for this issue.
Used SW version info:

[sharp.s1p.zip](https://github.com/mihtjel/nanovna-saver/files/4606272/sharp.s1p.zip)
[znc3.s1p.zip](https://github.com/mihtjel/nanovna-saver/files/4606275/znc3.s1p.zip)
[nanovnasaver.zip](https://github.com/mihtjel/nanovna-saver/files/4606278/nanovnasaver.zip)
[image]

Desktop sw: Windows10

Many thanks in advance!

MZachmann, 2020-06-29:
Can you please describe in more detail what issue you are seeing between the various S1p files? Do note that nanovna-saver seems to save as RI (real, imaginary) while sharp is using MA (mag, angle). I see differences in the data but they don't seem substantial.

Herman-RFguy, 2020-06-29:
Good evening.
thank you for following up my remark which i placed some time ago.
At first i want to express my appreciation to all those people working on this really superb software!

I spent some hours today to make a recap of my findings.
After finishing this i have to conclude i was wrong.
I do apologize for that. 
I add a short note i made today to investigate again how to calculate from s1p in format real + imaginary.
I must have been wrong last time in my calculations.
We can therefore close this issue.

Again, i like saver very much for the many features it has.

I have another question
I made a calculation scheme to determine the complex µ for toroids.
It works very well and requires a special home made test device.
when this calculation scheme is implemented in the saver software than the µ' and µ'' can be graphically displayed
just like the Q of a coil already can.
If there is interest in this i am happy to support with the know-how for that.
I realized it in excel now and graph with another program.
works fine.

With kind regards,
Herman,
PA0VRE

[image]

zarath, 2020-06-30:
Of course allways interested in such functionality..
Can you create a feature request with the needed informations?

closed 2020-06-30 by zarath