NanoVNA-Saver/nanovna-saver #182: ASCII Decode issue

abc123me, 2020-06-08:
**Describe the bug**
NanoVNASaver fails to read the firmware from the NanoVNA

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in upgraded 1MHz to 1500MHz VNA
2. Press connect in NanoVNASaver
3. Software crashes

**Expected behavior**
It to connect and work like normal

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Linux 5.4.36-1-MANJARO
 - Python version: Python 3.8.3
 - Saver version: 0.2.2

glovepost, 2020-06-09:
Ran into the same issue today
[image]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 2: ordinal not in range(128)
Aborted (core dumped)

NanoVNA-H
version: 0.4.5-4-2-geb1a66b

abc123me, 2020-06-09:
Looks like it comes from the same `readFirmwarer` function too

abc123me, 2020-06-09:
Looks like it came from the same function too (Sorry for closing the issue earlier - meant to click comment + new high dpi mouse)

glovepost, 2020-06-09:
I suspect that it might be a firmware issue, what version are you running on the NanoVNA?

https://groups.io/g/nanovna-users/topic/nanovnasaver_runtime_fault_on/74674213?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,74674213

abc123me, 2020-06-09:
NanoVNA-H version `0.4.5.1-gfbbceca` built `Dec. 26th 2019 - 19:32:31`

zarath, 2020-07-05:
Stil valid with current firmware?

abc123me, 2020-07-06:
Same problem, I just compiled and tested the most recent version of the firmware on [https://github.com/hugen79/NanoVNA-H/](https://github.com/hugen79/NanoVNA-H/)

abc123me, 2020-07-06:
OK, I done some investigation of my own and when I first connected to the NanoVNA-H on the serial port it outputted lots of bullshit, this random bullshit is causing the NanoVNASaver program to quit, I verified this fixes the bug by first connecting via picocom then connecting with NanoVNASaver.
[image]

zarath, 2020-07-06:
Can you please try with v0.3.4 or even current Development branch. I sadly did not own a -H variant.

abc123me, 2020-07-07:
There is no v0.3.4 for the NanoVNA-H, it uses custom firmware from https://github.com/hugen79/NanoVNA-H/releases, I'm on the latest v0.5.0 release for the NanoVNA-H

zarath, 2020-07-07:
v0.3.4 of NanoVNA-Saver I've ment

abc123me, 2020-07-07:
Same problem:
[image]

zarath, 2020-07-07:
are you doing source installation?
Then 
https://github.com/zarath/nanovna-saver/tree/development
would interest me. 

It reads up to 64k of data if it seems weird at port open

abc123me, 2020-07-07:
Yeah, I cloned and checked out the dev branch from https://github.com/zarath/nanovna-saver.git and same problem:
[image]

zarath, 2020-07-07:
can you provide me the debug file?
./nanovna-saver.py -D debug.txt

abc123me, 2020-07-07:
Dafuq? It works the debug file enabled! Here is the debug file: https://pastebin.com/YJFqbYCC
[image]

zarath, 2020-07-07:
Argh.. Perhaps a weird ass timing problem... you can try to modify wait default in NanoVNA/Hardware/VNA.py:exec_command function
or in somewhere in Hardware.py in the detect_version between write and read ....
But just guesses

abc123me, 2020-07-07:
Yeah, ill look into it whenever I get some free time. I can call `./nanovna-saver.py -D /dev/null` for now

ehaag, 2020-07-23:
I'm really happy to see some activity in this project again, as I've been having the same problem for a few months now. Running Manjaro 5.4.52-1 and Python 3.8.3. The rest is slightly different, with original NanoVNA hardware and upgraded firmware version 0.7.1-7-gfb57511 from edy555/ttrftech, built June '20. NanoVNASaver is version 0.3.6, released 6 days ago from the looks of it.

Fails in the same way, with a UnicodeDecode/ASCII codec error upon initiating a connection with the NanoVNA. However, unlike abc123me I am unable to get the connection to succeed even when using debug mode. I've tried with output to the console, to a text file, and to /dev/null with similar results. Sometimes it will hang indefinitely when connecting and I'll have to ^C out of it, other times it will immediately crash.

cropinghigh, 2020-09-30:
Having same problem on Manjaro 5.8.6, Python 3.8.5, NanoVNA-H v3.4, latest saver version(from source). Not helping both of "clearing" port with picocom or running in debug mode. 
Debug log:
```
2020-09-30 20:02:02,669 - NanoVNASaver - INFO - Startup...
2020-09-30 20:02:02,734 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-09-30 20:02:02,734 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-09-30 20:02:02,735 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-09-30 20:02:02,735 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (3600000, 30000000)
2020-09-30 20:02:02,735 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-09-30 20:02:02,778 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-09-30 20:02:02,981 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.7"
2020-09-30 20:02:02,981 - NanoVNASaver.Windows.About - INFO - This is 0.3.7-rc05
2020-09-30 20:02:02,987 - NanoVNASaver.Windows.SweepSettings - DEBUG - update_band(False)
2020-09-30 20:02:03,030 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 167.000000, normal font: 167.000000, factor: 1.000000
2020-09-30 20:02:03,073 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port /dev/ttyACM0
2020-09-30 20:02:03,077 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-09-30 20:02:04,176 - NanoVNASaver.NanoVNASaver - INFO - Connection /dev/ttyACM0 (NanoVNA)
2020-09-30 20:02:04,385 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-09-30 20:02:04,508 - NanoVNASaver.Hardware.Serial - WARNING - unable to drain all data

```

Exception:
```

2020-09-30 20:02:04,508 - NanoVNASaver.Hardware.Serial - WARNING - unable to drain all data
Traceback (most recent call last):
  File "nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 510, in serialButtonClick
    self.connect_device()
  File "nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 532, in connect_device
    self.vna = get_VNA(self.interface)
  File "nanovna-saver/NanoVNASaver/Hardware/Hardware.py", line 83, in get_VNA
    vna_version = detect_version(iface)
  File "nanovna-saver/NanoVNASaver/Hardware/Hardware.py", line 117, in detect_version
    data = serial_port.read(128).decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0x9b in position 0: ordinal not in range(128)

```

abc123me, 2021-01-25:
An update, I've found opening the device with picocom (serial terminal program, similar to minicom) and dumping any buffered serial data first solves the problem. When I open it I see a terminal full of jargon. I believe it empties some kind of jargon that the nanovna sends and is causing the decode to fail. After closing picocom and opening using nanovna-saver it works flawlessly.

closed 2022-11-10 by github-actions[bot]