NanoVNA-Saver/nanovna-saver #184: Crash when running continuous sweep

booyaamps, 2020-06-12:
Hello, just downloaded and built 0.2.2. After connecting to the nanovna, loading the calibration, and setting the sweep freq range (i've tried many different ones), i get the following traceback error:

Traceback (most recent call last):
  File "/home/jamie/nanovna-saver-0.2.2/NanoVNASaver/NanoVNASaver.py", line 755, in dataUpdated
    c.setData(self.data)
  File "/home/jamie/nanovna-saver-0.2.2/NanoVNASaver/Chart.py", line 3810, in setData
    self.calculateGroupDelay()
  File "/home/jamie/nanovna-saver-0.2.2/NanoVNASaver/Chart.py", line 3815, in calculateGroupDelay
    rawData.append(d.phase)
AttributeError: 'list' object has no attribute 'phase'
Traceback (most recent call last):
  File "/home/jamie/nanovna-saver-0.2.2/NanoVNASaver/SweepWorker.py", line 163, in run
    self.updateData(values, values21, i)
  File "/home/jamie/nanovna-saver-0.2.2/NanoVNASaver/SweepWorker.py", line 194, in updateData
    freq = self.data11[offset * segment_size + i].freq
AttributeError: 'list' object has no attribute 'freq'

Ubuntu 20.04, python3.8

booyaamps, 2020-06-12:
seems to be a dupe of #169. Closing this myself.

closed 2020-06-12 by booyaamps

zarath, 2020-06-12:
@booyaamps have you tried the Development branch or my fork?

booyaamps, 2020-06-12:
Yes, same issue. It works uncalibrated, but as soon as a cal is loaded, it
sweeps once and stops.

booyaamps, 2020-06-13:
in 0.2.3 development, here's the error it throws after loading a cal and trying to do a continuous sweep:

qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 1618, resource id: 13259650, major code: 40 (TranslateCoords), minor code: 0