NanoVNA-Saver/nanovna-saver #186: Installation on Debian 10 (buster) failed

googlegitter, 2020-06-16:
I tried to follow the installation description but it will not work.

First i have to install the following packages:
python3-pip python3-distutils python3-setuptools python3-wheel

Then i could start
```
# python3.7 -m pip install .
Processing /srv/bin/nanovna-saver
Collecting PyQt5 (from NanoVNASaver==0.2.2)
  Downloading https://files.pythonhosted.org/packages/8c/90/82c62bbbadcca98e8c6fa84f1a638de1ed1c89e85368241e9cc43fcbc320/PyQt5-5.15.0.tar.gz (3.3MB)
    100% |████████████████████████████████| 3.3MB 339kB/s 
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.7/tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-el3apevh/PyQt5/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-el3apevh/PyQt5/
```
that failed!

Afterwards i tried to manually install the packages one after another withe error messages trying to start "python3.7 nanovna-saver.py":
python3.7 -m pip install numpy
python3.7 -m pip install scipy
python3.7 -m pip install serial

But the last package could not be fixed:
```
$ python3.7 nanovna-saver.py
Traceback (most recent call last):
  File "nanovna-saver.py", line 17, in <module>
    from NanoVNASaver.__main__ import main
  File "/srv/bin/nanovna-saver/NanoVNASaver/__main__.py", line 23, in <module>
    from .NanoVNASaver import NanoVNASaver
  File "/srv/bin/nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 29, in <module>
    from serial.tools import list_ports
ModuleNotFoundError: No module named 'serial.tools'

# python3.7 -m pip install serial.tools
Collecting serial.tools
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/serial-tools/
```

Please - what mystic must be done to get it running?

googlegitter, 2020-06-16:
I could found the solution here: https://github.com/espressif/esptool/issues/350

So **python3.7 -m pip install pyserial** did fix it.
Now it starts. :-)

zarath, 2020-06-25:
also upgrading pip before installing should fix this

closed 2020-06-25 by zarath