NanoVNA-Saver/nanovna-saver #187: Error 2 While Installing

nokiaman2002, 2020-06-20:
I have all the pre-requisites installed.

Using Xubuntu 18.

When running python3.7 -m pip install .

I always get:

Processing /home/nokiaman2002/nanovna-saver
Collecting pyserial (from NanoVNASaver==0.2.2)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting PyQt5 (from NanoVNASaver==0.2.2)
  Downloading https://files.pythonhosted.org/packages/8c/90/82c62bbbadcca98e8c6fa84f1a638de1ed1c89e85368241e9cc43fcbc320/PyQt5-5.15.0.tar.gz (3.3MB)
    100% |████████████████████████████████| 3.3MB 358kB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.7/tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-dlbaux9z/PyQt5/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-dlbaux9z/PyQt5/

Has anybody any idea what this error could be and what is causing it?

Thanks in advance.

Gareth

zarath, 2020-06-20:
you pip is to old.

Try:
python3.7 -m pip install -U pip
to upgrade pip

or use a complete venv environment

nokiaman2002, 2020-06-20:
Many thanks for that, problem resolved!

closed 2020-06-20 by nokiaman2002