NanoVNA-Saver/nanovna-saver #189: Calculate the Common Mode Rejection (CMMR) of a Balun

satfan52, 2020-06-21:
reference:
http://www.arrl.org/files/file/QEX_Next_Issue/Nov-Dec_2010/Skelton_QEX_Nov-Dec.pdf

When measuring the performance of a balun with a single ended input and a balanced symetrical output, it is important to assess the Common Mode Rejection and not only the impedance transforming ratio or insertion losses. This is especially important for very wideband balun aiming at covering HF+VHF for instance.  Calculating the CMR of a balun requires to build a dedicated 3 port fixture (1 input and 2 outputs, see article) and to make two successive S parameter transmission measurements between the input and the two outputs.  

CMRR dB =20*log (S21+S31)/(S21- S31) 
(where port 1 is the input and port 2 and 3 are the two outputs) 

It would be great if Nanovna-saver could offer the possibility to display the CMR as one of the display parameters.  The idea being that nanovna-saver would use the two last values of S21 measured  to calculate the CMR based on the formula above

AD5U, 2020-07-12:
satfan52 - THANKS!  Thanks for including links to references in your post.  I read the article on the wideband balun with great interest as my friend and I have been researching making our own transformer+chokes for uses at HF, VHF and UHF bands.  I had expected that we would need to make multiple baluns for the various bands but that may change now.

Additionally we have been researching effective ways to test baluns and chokeswith our nanoVNA units (we love them!) and were thinking of fixture(s) that might be needed and we have more ideas now.  If we come up with things that actually work, we'll contribute.

The linked info also gave me more ideas for testing with some of the fancy analyzer(s) I use for work.  Some are programmable and/or remotely operable with data then made available to the computer for additional processing.  I might be able to work out the process and number crunching and then see what I can do with my nanoVNA units.  If I can, I will try to contribute something to the software project.

Thanks for the inspirations you have provided me!
Scott
AD5U

satfan52, 2020-07-18:
Hi Scott,

You are welcome.

 I have (not yet) tried to use the measurement techniques described in these two articles. If you do it before me, it would make sense to report here if the formula cited in the article (to measure the CMMR gives) good practical results or not.  

CMRR dB =20*log (S21+S31)/(S21- S31)

If not, there is no need to waste coding time on this. 

Another way to implement this with nanovna-saver that might be simpler to understand  for the end-user but a bit more challenging to implement for the coder, is  simply to provide the possibility in the FILE menu to "import an S2P file as  S31" . Then up to the user to choose display CMRR in the list of possible "display charts" and/or in the list of markers data.

closed 2022-11-10 by github-actions[bot]