NanoVNA-Saver/nanovna-saver #200: Can-not run v 3.2 under Ubuntu 18.04 LTS

satfan52, 2020-06-29:
Hello;

I have downloaded the "V3.0 executable" dedicated to Ubuntu from the "release page" and so far it seems to works very well. Many thanks  especially for having included the CH1 attenuation functionality !

However, I also wanted to test the "bleeding edge version" in the "master repository" (using a simple "git clone" command), but I got less lucky (see below)

root@on7yi-ubuntu:/opt/nanovna-saver# python3.7 nanovna-saver.py
Traceback (most recent call last):
  File "nanovna-saver.py", line 17, in <module>
    import pkg_resources.py2_warn
ModuleNotFoundError: No module named 'pkg_resources.py2_warn'
root@on7yi-ubuntu:/opt/nanovna-saver# 

I am confident I have installed all the dependencies as version 2.2 and 2.3 downloaded using a simple "git clone" commands are both running fine in my Ubuntu 18.04 LTS .  My version of python-setuptools is 39.0.1-2, I tried to reinstall it but it does not make a difference. So I am a bit lost

zarath, 2020-06-29:
Try to update setuptools ..
python3 -m pip install --upgrade setuptools

satfan52, 2020-06-29:
Thanks, but upgrading to setuptools-47.3.1 does not solve the issue unfortunately.

satfan52, 2020-06-29:
Thanks!

- python3-pkg-resources is already the newest version (39.0.1-2), so it did not help 
- but commenting line 17 as indicated below solved the issue
**#** import pkg_resources.py2_warn  
Regards

closed 2020-06-29 by satfan52