NanoVNA-Saver/nanovna-saver #202: Binary Release 3.0 - Ubuntu 18.04 LTS -  s2p files export - erronous S21 data

satfan52, 2020-06-29:
Hello,

Two points

1) after exporting S21 files in SimSmith, I noticed that S11 data is OK but S21 is not, there is obviously something very wrong with the export of s2p data
2)  I noticed that that the data exported for S21 is the same independently of the attenuation setting chosen for CH1. I am no sure if this is by design or not but this seems to me inadequate. After 1 is fixed,  it would be great that s2p data exported takes into account the effect of the CH1 attenuation 

See pictures below, the first one shows that the attenuation settings works well (40dB attenuation), the second one shows that the S2p data exported is not corresponding to the measurement 

[image]

[image]

zarath, 2020-06-30:
Yes there was a bug in applying data not to s21 in calibration.
Should be fixed in current development branch.
But no fresh binaries yet

zarath, 2020-06-30:
See v0.3.3

closed 2020-06-30 by zarath