NanoVNA-Saver/nanovna-saver #203: Assistant function to guide full two port measurements and create fully populated s2p file with 4 S parameters.

satfan52, 2020-06-29:
When I need to make full two port measurements, typically for measuring non-symmetrical devices such active components wityh the objective of making CAD simulations, I am doing two distinct sucesive measurements;
-  the first for S11 and S21 and,
-  the second (after reverting CH1 and CH0) for S22 and S12.
 I then merge the two corresponding s2p files into a single one with the 4 S parameters (8 numbers) using Excell.

This process is slow, cumbersome and very error prone, not user friendly at all and for that reason I am not using it as often as I should. The idea is to have an assistant function that is similar to the one that is available for SOLT calibration. The assistant would ask to make the two measurements in an orderly way  and at the end it would create the S2P file with the 4 S parameters. (Exporting it would be done the usual way). 

If nanovna)saver

satfan52, 2020-07-18:
Another simpler way to avoid modifying the assistant function is simply to add a check box in the file export menu. When the check box is checked, the exported S2P file is actually a combination of the two last transmission measurements. So instead of puting 0s for S12 and S22 in the exported S2P file, it would populate S22 and S12 with the data from the PREVIOUS sweep so the S2P file exported is a full blown one with the 4 S parameters and not only S11 and S21. Of course, up to the user to know what he/she is doing when he checks that box next to the export S2P button

satfan52, 2020-08-27:
The nanovnaQT software of the nanovna V2 includes that functionality

satfan52, 2021-03-13:
This functionality is critical, without it nanovna-saver can not easily export the 4 S parameters in an S2P file.

rjordans, 2021-06-22:
I typically use a short python script for this using scikit-rf that at least takes some of the possible errors out:

```python
import skrf as rf
rf.stylely()

# Load partial networks
mrf = rf.Network('input-side.s2p')
tmp_output = rf.Network('reverse.s2p')

# Copy data from the reverse measurement into the model
mrf.s[:,0,1] = tmp_output.s[:,1,0]
mrf.s[:,1,1] = tmp_output.s[:,0,0]
mrf.name = "MRF9060 @ 200mA"

# Plot figure as sanity check and save to new file
mrf.plot_s_db()
mrf.write_touchstone('mrf9060.s2p')

```

zarath, 2021-06-22:
what about an option "switch with S12/S22 data"

satfan52, 2022-07-01:
Hello,
It has been a while I checked this repository. Has this functionality been finaly implemented so nanovnasaver can be used as a true vna and not a simple antenna analyzer ? If yes, I will close this ticket. If not,  any implementation will do the job provided that an S2P file can be exported with the 4 S parameters and not just S11 and S21.  I can test nanovnasaver any longer as it sadly does not support my nanovna v2+4

KJ7LNW, 2023-02-15:
Here is a Perl example based on `RF::Component` that will do it.  It can rescale frequency mismatches between the input files, too, using simple linear interpolation:

```
./s2p-merge.pl file1.s2p file2.s2p output.s2p
```

```perl
#!/usr/bin/perl

use warnings;
use strict;

use Getopt::Long qw(:config bundling);

use PDL;
use PDL::IO::Touchstone qw/pos_vecs_to_m/;
use RF::Component;

my %opts;

GetOptions(
	"freqs|f=s"		=> \$opts{freqs}
);

if (!@ARGV)
{
	print STDERR "usage: $0 [--freqs F] file1.s2p file2.s2p output.s2p\n";
	print STDERR "Merge file1[S11,S21] and file2[S11,S21->S22,S12] into output.s2p.\n";
	exit 1
}

my ($fn1, $fn2, $out) = @ARGV;

my $c1 = RF::Component->load($fn1);
my $c2 = RF::Component->load($fn2);

if ($opts{freqs})
{
	$c1 = $c1->at($opts{freqs});
	$c2 = $c2->at($opts{freqs});
}

die "freqency mismatch" if (any $c1->freqs != $c2->freqs);
die "z0 mismatch" if (any $c1->z0_ref != $c2->z0_ref);

# Row-major
my $out_S = pos_vecs_to_m(
	$c1->S(1,1), $c2->S(2,1), # S11[c1:S11] S12[c2:S21]
	$c1->S(2,1), $c2->S(1,1)  # S21[c1:S21] S22[c2:S22]
	);

my $cout = RF::Component->new(
	freqs => $c1->freqs,
	S => $out_S,
	n_ports => 2,
	z0_ref => $c1->z0_ref,
	comments => ['Merged files: ', $fn1, $fn2]
	);

$cout->save($out, output_fmt => 'RI');
```

closed 2023-06-17 by github-actions[bot]