NanoVNA-Saver/nanovna-saver #206: Measuring µ' and µ'' with nanovna-Saver

Herman-RFguy, 2020-06-30:
Dear members,
This is the first time i contribute an idea to this community.
I developed a measurement method to determine the complex µ ( µ' and µ'' ) of a toroid using the nano vna.
after a calibration sequence the S11 is saved in a .s1p file and than the physical dimensions of the measured toroid are added to the excel sheet i made.
I plot the numerical results from the excel sheet with the use of a plot program i found on the WWW.( see pic.)
The measurement chamber can be easily constructed with double clad PCB, or some copper plate, the round wall is not obligatory as tests learned, using 4 spacers yields identical results. More mechanical details can be given in a separate document.
For saver:
The physical dimensions of the toroid have to be input, the s11 data is obtained directly from the vna.
After ending the calculations the µ' and µ'' are known and can be graphically shown in the plot window of saver.
The frequency axis is preferred logarithmic and the Y axis as well, auto ranging as µ can be 10 for a powder iron core
and >1000 for other types.
I have done all the math and i am happy to share this with the community to add a very nice function.
The AL value can be derived easily and it is used very often for calculating the number of turns for a self inductance.
A plotted result for a T94-2 toroid is added to show what can be expected.
µ'' tells a lot of the losses and thus enables choices to be made for the intended use and frequency.
I wait for any help how i have to to share this math info .
[image]
[image]
[image]

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

zarath, 2020-07-14:
Can you attach the excel-sheet or the formulas to this issue?
For the construction details of the chamber.
It looks like that the inner rod is about 4mm diameter brass, soldered to the bottom plate.
On the top of the rod there seems to be soldered an inner jack of a BNC connector, so i guess
the is a plug pin added under the BNC-Jack on the top plate, right?

Herman-RFguy, 2020-07-14:
Hello Zarath,
On the top of the rod is a inner jack of the BNC connector.
This is the inner jack of the BNC connector soldered on the top plate.
It has to be removed from the BNC connector first and the inner teflon of the BNC has to be made just a little bit wider so that the inner jack goes in without using force.
On the bottom the rod is soldered.
In this way the DC resistance is reduced to a minimum.
The rod is app. 4mm in diameter.
Tests has shown that the round wall may be omitted when 4 vertical spacers are used, it yields the same results and it makes the construction more easy .
[mu_calculator_saver_V13.xlsx](https://github.com/NanoVNA-Saver/nanovna-saver/files/4919822/mu_calculator_saver_V13.xlsx)
I will add the formula part in one hour from now as dinner is served right now.
I have a power point presentation about this topic, it is in Dutch but all the math is in it.
I add it together with the math.
You are most welcome with any questions,
73,
Herman,
PA0VRE
[image]

[µ meting met nanoVNA-V4.pptx](https://github.com/NanoVNA-Saver/nanovna-saver/files/4919829/meting.met.nanoVNA-V4.pptx)

zarath, 2020-07-14:
Many thanks!

Herman-RFguy, 2020-07-14:
Hello Holger,
If you have problems with reading Dutch, let me know, i will translate in German or English depending on what you prefer.
I am interested in following the proces, please keep me posted.
Thanks,
Herman.

Herman-RFguy, 2020-07-14:
one more comment:
In formula (3) and (4)  N square  = 1
because of the used measurement chamber.
This measurement chamber is a standardized method used by all manufacturers.

radi8, 2020-07-15:
Hello Herman,

I tried using nanovna-saver sweeping 50KHz to 200MHz with segments =1 for 101 points. My resulting steps are linear at 2MHz per step but I see that yours are (probably) logarithmic which is much more ideal. Could you please explain how you did this as I can not see any setting for a logarithmic sweep. I have not made a chamber and am just operating with a single turn of wire in an attempt for a quick and dirty reading whose accuracy does not need to be very good simply enough to approximately identify the material type. Perhaps you can warn me if this will not be good enough for this purpose and I will need to make a chamber even with the four mounting posts only.

73, Graeme ZL2APV

Herman-RFguy, 2020-07-15:
Hi Graeme,
Nice to see that nanovna is popular at the other side of the world as wel!
I did make a plot with a logarithmic frequency axis but used a data set obtained with nanovna-saver.
That means that the datapoints are equidistance in the frequency domain as you mentioned, in your case 2 MHz spaced.
plotting  this date set with a logarithmic frequency scale has a disadvantage.
Each decade the number of data points is a factor 10 higher than the previous one.
Plotting 3 decades and 101 datapoints means that the distribution of datapoint is 1:10:1000 in the 3 decades.
In other words, the lowest decade has 1 the following 10 and the last one 89 points.
To fix this i recommend to do a separate sweep in the lowest decade and merge this data with the results of a second sweep covering the other decades.

For Holger:
If you can fix this in some way in the application for µ measurement that would be a valuable addition.

I hope this answer helps you Graem.
All the best,
73,
Herman,
PA0VRE.

zarath, 2020-07-15:
Logarithmic segment scanning would be a possible option. So when 3 segments from 1M-100G first scan points would be from 1M to 10M, second from 10M to 100M and third from 100M to 1G.

Logarithmic sweeping is imho not impemented in any current firmware. Perhaps scan point reduction can spread more evenly without increasing scan time too much.

Herman-RFguy, 2020-07-15:
Hi holger,
It is good to mention this detail.
An approach could be to scan 101 points in each decade, equidistance in frequency.
Plotting so obtained data yields a pretty good result and is, i assume, the most convenient way to do as this does not change the sweep algorythm, start & stop frequency are adapted for each decade and s11 results stored in one array.
It is just a thought i have, other ideas will work too, it is the sw writer's creativity and freedom.

radi8, 2020-07-15:
What I have done is swept from 50KHz to 200MHz with 20 seg and saved the .s1p followed by sweep at 2 segments again saving and finally at 1 segment. I took the values from 50KHz to 999KHz from the 20 seg file added to them the 1MHz to 9.9 MHz from the 2 seg file and finally 10MHz up from the 1 seg file. I had to drop off the last 13 readings for 101 points so only went to 174 MHz but at least the pseudo log file worked and only took a couple of minutes to create. I just need to sweep a bit higher if I want to go up to 200MHz.
Still getting inconsistent results with my FTnn-43 toroids so I will lash up a chamber with heavy wire standoffs holding 2 PCB plates apart and a centre wire which I will unsolder from the connector to slip the toroid on and then solder back to see if that is better than 1 turn.

radi8, 2020-07-16:
I constructed a chamber or cage to measure the problematic FT240-43 toroid and got a huge improvement in the result. The cage was constructed from 2 pieces of PCB cut to 70x70mm with copper wire spacers and the centre wire soldered to bottom plate with toroid slipped over and soldered to sma connector. Without the toroid the cage sweeps as an inductor following the outside of the Smith chart. The graph attached is from your spreadsheet with my psuedo log input file from a .s1p compilation.

I tried an FT50-43 toroid in the cage but real and imaginary didn't cross over so I think the cage needs to be sized to suit the core being tested. I am still experimenting and maybe I can achieve similar results by varying the turns around a wound toroid and building up a catalog from that.
[image]
[image]

Herman-RFguy, 2020-07-16:
Hi Graeme,
Nice!
good to see the improvement.
The heighth of the chamber could be reduced to minimize parasitic self inductance. I measure 33nH of my chamber.
concerning calibration : open short and load is the standard procedure for calibrating.
Connect the chamber instead the short and the results  for low µ toroids will show smaller errors.
This still implies a systematic calibration error but taking into account that the tolerance of the µ as specified by the manufacturer is +- 20%  this is neglectible.

For the nerds reading this:
 the perfect way to calibrate eliminating the parasitics of the chamber is
1
calibrating at the end of the cable with open short and load
2
measure & save the s11 of the empty chamber for all frequencies used in the measurement
3
measure & save the s11 of chamber + toroid
4
devide s11 obtained in 3 by s11 obtained in 2
5
the resulting s11 data obtained by the division in 4  is the data used to calculate µ' and µ''

I do not recommend this as the accuracy obtained in this way does not make sense in relation to the accuracy of the manufacturers specs of the toroid.
The choice is yours!
73,
Herman,
PA0VRE

Herman-RFguy, 2020-08-20:
Hi Holger,
I translated my presentation into english and edited the formula work in Word, making it much more readable compared to my handwriting -:)
I attach it hereby.
do you plan to implement this nice feature when you fint the time for it?
With kind regards,
Herman.

[µ meting met nanoVNA-V4-E.pptx](https://github.com/NanoVNA-Saver/nanovna-saver/files/5101923/meting.met.nanoVNA-V4-E.pptx)

zarath, 2020-08-20:
@Herman-RFguy tnx for the update. I still plan to impelent that functionality, but time...

At least I have build a test facility already.
[image]

Herman-RFguy, 2020-08-20:
Great to read that Holger!
Your chamber  looks great!
If i may give you a suggestion:
The height of the chamber is rather high, creating much self-inductance.
Mine has 33nH for instance.
What measures yours?
See my discussion with Greame, he made it high too, and after reducing the height the results were better.

We can't buy time, its precious, so if you find some i am looking forward to be your beta tester.

with kind regards,
Herman.

zarath, 2020-08-20:
It's build after that original dimensions in the presentation. I haven't measured the exact induction yet. Reducing the hight wouldn't be easy afterwords as   I brazed the stoods.

Herman-RFguy, 2020-08-20:
never mind Holger, it will work anyway.
You will see a little error when measuring low µ toroids, in the lower frequency range, thats all.

radi8, 2020-08-21:
Hi Herman,

Many thanks for the English translation and it is a big help. Still 
getting excellent results from my lashed up test facility and it has not 
encouraged me to make a better version although the one Holger has built 
looks similar to what I had in mind but I will probably see how 35mm 
spacers will work as I have a set I made a couple of years ago. I keep a 
selection of N and BNC parts so will have no problem in building a 
mating connector. I will use knurled nuts on the top of the pillars 
which have 3mm threads projecting.

73, Graeme

igor-m, 2020-09-23:
FYI - I've been using a much simpler approach - just 2 turns of wire via the toroid connected to S11 input.
Using 5 lines with math on s1p data in gnuplot.
[image]
[image]

Herman-RFguy, 2020-09-24:
Thanks for your contribution Igor-M.
Using 2 windings will do for low frequencies but the parasitic capacitance between the windings wil give erroneous data for higher frequencies.
( Why your plots ending at 30MHz? )
another issue with windings is the repeatability.
The windings are never the same for each toroid as they are hand-wound, introducing an uncertainty in the results.

Concerning the math:
All lines given in my scan show the algebraic derivation which results in formulas (3) and (4)
The real and complex coefficients of S11 and the physical dimensions of the toroid and the frequency related to the complex coefficients are input in this formulas, so just one line of math for µ' and one for µ''

I gave the derivation just to explain the background of the calculation.
 73,
PA0VRE

radi8, 2020-09-24:
Thanks Igor-M for your interesting work. It is always good to see another approach. I will be staying with what I have as it works so well for me and is repeatable but I accept that I did need to make the chamber which was not too difficult and only took around an hour to make. I do have to solder and unsolder the centre wire to put the toroid on and off but it is something that I don't do very often so its not really a problem. I massage the .s1p file so that I can directly import it into Libre Office spreadsheet and the graph is directly displayed using a near copy of Herman's sheet.

73, Graeme ZL2APV

igor-m, 2020-09-24:
@Herman and Graeme - your hardware is definitely better than my 2 turns of wire, sure!!
I've seen a paper in past with the copper chamber as the Herman's one, where they took the size of the chamber and the size of the toroid into the calculation as well.  I've tried the simplest way - I started with 1 turn (used a 2.5mm dia shielding from an rg178 teflon silver coated coax as the wire) then 2 turns and 3 turns. The 2 turns matched the datasheet curves best so I stayed with 2 turns. I am not extremely interested in absolute values (therefore I do not care on the toroid size), but more in the actual "shape of the u' and u'' curves" as it indicates the material pretty well. Why 1-30MHz? I want to measure some junkbox toroids for my EFHW antennas in 80-10m range :)

igor-m, 2020-09-24:
Here is the link to the paper 
[https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjpyuSljYLsAhUMlxQKHfaYA24QFjAAegQIAxAB&url=http%3A%2F%2Fprzyrbwn.icm.edu.pl%2FAPP%2FPDF%2F117%2Fa117z522.pdf&usg=AOvVaw0AN7q-fldEfPZfic10o1a2](url)

igor-m, 2020-09-24:
@ Herman - out of curiosity I've put my above s1p (T50-6 with 2 turns, 1-30MHz, 801 points) data into your excel table. I've added the "Turns Number" to the calculation (.. *n*n in the L column).
FYI - here is your excel with my data - I get even higher numbers that the mine one above. What could be the issue?
[mu_calculator_saver_IGOR_M_FT50_6.xlsx](https://github.com/NanoVNA-Saver/nanovna-saver/files/5278125/mu_calculator_saver_IGOR_M_FT50_6.xlsx)

Herman-RFguy, 2020-09-24:
Hello Igor,
Interesting tests you are doing.
and yes you are right if you are interested in the behavior of the junkbox toroids, the curves tell you if they are suited for the HF frequencies, good job!
I will have a look to your excel and come back to you.

With kind regards,
Herman.

igor-m, 2020-09-25:
@ Herman - I've made 8 turns with my powder T50-6 core (u=8 from datasheet).
Below my measurement and the same data in your excel. Mine looks "feasible", your a bit higher than expected.
[image]
[image]

Herman-RFguy, 2023-02-10:
Hi Zarath, it still has your interest?
Regards,
Herman

rjordans, 2023-02-10:
This sounds like a great addition indeed.  We already have a chart that can plot Z in real and imaginary parts (Charts/RI.py).  Maybe adding a derived variant of that with some additions to the drop-down menu to set core dimensions, number of windings, and adjusting the calculation in getImYPosition() and getReYPosition() could be a quick fix?

Looks doable to me but I'm afraid that I don't have time to make it happen soon.

Cheers,
 Roel (PA3RJ)

rjordans, 2023-02-10:
Another thing though, considering the measurement itself:

VNA accuracy is best when
- measuring reflections when they are small (S11 of something close to 50 ohm) or
- when measuring something with attenuation (S21)

Measuring a large inductor (or a small one at high frequencies) gets you quite a bit away from the 50 ohm.  Might be worth an experiment to make a new box with a connector on both ends of this central rod and using a S21 measurement. This puts the inductor in series with the measurement path and will cause a large attenuation for a large impedance, making the measurement much more accurate.  NanoVNA-Saver already implements the calculation into Z-series so that may give you a nice start on the calculations.

Cheers again,
 Roel

rjordans, 2023-02-11:
Ok, this got stuck in my head so I gave it a go (code available on [my fork](https://github.com/rjordans/nanovna-saver/tree/main)).

I've added a graph based on the existing real-imaginary graph.  The settings for the core parameters are in the graph menu (right click on the graph and you'll find them).

I don't have a nice test fixture but I did a quick test on some unknown quality small ferrite torroids I had laying around 8mm OD, 4mm ID, 3mm height.  Example setup (small core around a wire soldered to an SMA connector.  Sweep 10M-1G in 3 segments.

[image]

RI plots already supported a log frequency scale.

[image]

Corresponding |Z| measurement
[image]

Looks like it works, I'm not really happy though with how the core properties are set.  Might be nice to have a single, more flexible dialog window where you can specify the dimensions in different styles depending on your core type and setup.  Maybe even with a list to select from some standard core dimensions.

Anyway, I don't have the nice test setup or good reference cores here so it would be great if someone can test this code already.

Cheers,
 Roel

Herman-RFguy, 2023-02-11:
Hi Roel,
I am really excited about your work!
Just downloaded your zip file of Saver with this extension.
I will explore this version and let you know my results asap.
W. kind regards,
Herman.

Herman-RFguy, 2023-02-11:
oops, i thought i would find a Windows executable as I am used to, but it is a Pyton source code.
I am not familiair with that.
Can you provide me with an executable Roel?
Thanks in advance,
Herman

rjordans, 2023-02-11:
Hi Herman,

Ah, that's to bad.  I'm not sure how that works.  I've only got a Linux machine here...  I see though that @zarath has added a script to have Github run the build process for releases.  I've kicked that off so hopefully there will appear some usable version for you on my fork.

Well, it seems to have done something at least.  At the bottom of the page linked below there are some outputs.

https://github.com/rjordans/nanovna-saver/actions/runs/4151729912

Cheers,
 Roel

Herman-RFguy, 2023-02-11:
Thanks a lot so far Roel.
I keep an eye on it and will report my results.

w. regards,
Herman.

Herman-RFguy, 2023-02-11:
Hi Roel,
I have downloaded a working version!
Im am busy doing reference tests with this sw and my hand-made results.
first impression:
It seems to work!!
te be continued

Herman-RFguy, 2023-02-11:
Hi Roel,
I tested 2 cores i had measured by hand with my excel sheet.
The plots compared to nano-vna-Saver are good.
some differences with the big core can be explained as follows: i have more then one core and i don't know wich one u used -:)
My compliments to the men who worked on this.
The core dimensions can be entered by right clicking the mouse while hovering over the plot area of the screen.
This has to be explained to the users to enable them to discover this very useful feature.
Now you are able to judge the junkbox cores and much more....
With kind regards,
Herman.
[image]

[image]

rjordans, 2023-02-12:
Great to hear Herman!  Good to know that this produces usable results.

I've added a brief description on how to use this feature in the toplevel README.md document (also visible on the main Github page).  Hopefully that's enough for people to understand the basic usage.

73 de PA3RJ, Roel

Herman-RFguy, 2023-02-12:
I think so Roel.
I will write an article to be published in some ham radio magazines with some explanation how to use this feature and interprete the  curves.
You are doing a great job in your leisure time by contributing your skills to this kind of projects.
Best 73,
Herman,
PA0VRE
 (by the way, my doughter also works on Radboud university neuro sciences)

Herman-RFguy, 2023-02-12:
Hi Roel,
I found a bug.
Regardless the choosen function displayed by a trace when you stop the sweep mode the application freezes and has to be started again.
This needs a little tlc  .
w. regards,
Herman.

rjordans, 2023-02-12:
Hi Herman,

Is this bug something that this new code triggers or is it something else?  I see that there is another issue #589 that there is a problem in general with the continuous sweep not stopping.  You might be running into that?

Cheers,
 Roel

rjordans, 2023-02-12:
@Herman-RFguy this build https://github.com/rjordans/nanovna-saver/actions/runs/4157198674 should fix your freezing problem hopefully

Herman-RFguy, 2023-02-12:
Thanks a lot Roel!
I let you know in the coming days.

Herman-RFguy, 2023-02-12:
I did not notice your previous question.
It might be as well the case that i run into that issue Roel.
Than it has no root cause in the u' and u''  addition.
It is not possible for me to determine that but it is likely that it is an already existing issue.
Cheers,
Herman.

Herman-RFguy, 2023-02-12:
Hi Roel,
I just tested the version that repaired the freezing bug.
It does not freeze anymore as i stop and start the sweep mode.
Compared to the previous version a new bug occurred, i can not enter the core area and effective length anymore.
Murphy is here.
greetings,
Herman.

rjordans, 2023-02-15:
Hi Herman,

That's strange.  I didn't touch any parts related to the mu calculation for fixing the freezing bug and can see no change in the behavior here either.

Cheers,
 Roel

Herman-RFguy, 2023-02-19:
Hi Roel,
I did some additional testing and found out that the prerequisites for changing this parameters are important.
Stopping the sweep and than entering the phisycal parameters of the toroid works.
in other words: you have to learn the "habits" of the sw to work succesful with it.
The over-all impression is that saver is not very robust in this phase and easy ends in a hang-up.
The only way is sometimes restarting the PC as it won't start after a hang-up any more.
That is not a u measurement related topic i assume but is maybe known to the community already?

what is desirable now?
Do you want me to investigate " bad habits and report them to you?
I am willing to help you and others to make this a succes. ( it already is to my opinion!)

with kind regards,
Herman.

igor-m, 2023-02-21:
Hi, where one may download the version with mu for testing? I can see rjordans built above, but it cannot be downloaded..

Herman-RFguy, 2023-02-22:
Hi Igor,
6 messages above yours is a hyperlink to the last build.
scrolling downwards shows the link ( X64) you can download
Your feed back is much appreciated after testing.
It works very nice, do not forget to enter the physical dimensions of the toroid under test by right clicking on the graphic output window.

Herman-RFguy, 2023-03-12:
Hello Roel,
I have 2 improvements which are very useful for a new release.

1)
The autoscaling of u' and u'' 
Use one scale value for both axes this will be u'
This is very desirable as you want to compare both values  which is much more convienent with one scaling factor.

2)
Can you also plot in the same display the quotient u''/u' as this represents the loss tangent of the core material and gives very useful information on selecting the frequency where to use the core material for what purpose.

Thanks again for your contribution for this topic.
Today i gave a webinar ( on RF-seminar) and there were 185 interested listeners.
This topic is much appreciated by many people
with kind regards,
Herman.

rjordans, 2023-05-08:
Hi Herman,

I haven't had much time to look into this but maybe something to note on your point 2:  we already have a S11 Q plot that should basically show u'/u''.  Maybe that's enough for you to work with already?

Implementing the common scaling for u' and u'' looks like it will be a bit more involved than what I was hoping for.  I like the idea but it will take some time before I'll be able to look into that.

Cheers,
 Roel

Herman-RFguy, 2023-05-08:
Hi Roel,
Regarding 2: I think so, i will check this and let you know.
Regarding 1: The same scaling is very desirable and i understand it is all done in your leisure time.
I appreciate very much  what is already done.
I received much positive response on the addition of this u' & u''  feature as it saves much work to generate this graphs.

I saw your qrl activities, very interesting.
Thanks again and when you have time i notice your message on this medium and will react.
With kind regards,
Herman.

PR #595 (2023-02-19): Measuring mu' and mu'' of ferrite cores
This change adds a new plot type "S11 mu" to show the complex permeability of an inductor core.

Cores are measured by putting one or more windings on them and measuring the S11.

In order to correctly calculate the permeability the user also needs to set the dimensions of the core (cross-section and effective length) and the number of windings.  This can be done by opening the context menu for the plot (right click on the graph area).

The core dimensions can either be measured from the core directly using calipers.

More info on how the calculations wo [... cut]