NanoVNA-Saver/nanovna-saver #210: App size is too large in Win10

imreac, 2020-07-01:
When opening the latest version of NanoVNA Saver 0.3.2 in Win10, the application is too large to fit in one window view.  Whether full screen or not, the app will not fit.  Tried reducing text size, which makes a small difference, but always need to scroll around program window to see everything.  
**To Reproduce**
Steps to reproduce the behavior:
1. Click on desktop Icon to open NanoVNA Saver 0.3.2
2. App opens, but does not fit in window
4. See Screenshots
**Expected behavior**
Expected app to fit in window.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Win 10   (see screenshot)
 - Python version: [if applicable]
 - NanoVNA-Saver version: [e.g. 0.3.2]

**Additional context**
Deleted .ini file here: C:\Users\Admin\AppData\Roaming\NanoVNASaver   Still no difference.
Desktop Resoloution:
[image]

On very first startup, had some message in command window to the effect that " could not get desktop resolution..."

imreac, 2020-07-02:
Fixed issue by resetting Windows Scaling from 150% to 100%.  This unfortunately shrinks Tabs and everything else (i.e. file manager folders)
See attached screen captures:
[image]
[image]

RandMental, 2020-07-02:
When there are only 4  graphs, the desktop scale correctly on the screen (1680x1050 display)
When two more are enabled e.g. all 6, the screen-width becomes too long.
The aspect ratio might need to change

[image]

imreac, 2020-07-02:
I had not tried this, but it needs attention.  Even if the additional 2 graphs (#5, 6) are closed, the original ones on the right  (#3, 4) are stretched.
[image]

closed 2022-11-10 by github-actions[bot]