NanoVNA-Saver/nanovna-saver #211: Wrong s2p file generation

KobaSisyo, 2020-07-01:
**Describe the bug**
The S21 data in the generated s2p file appears to be S11.

**To Reproduce**
Steps to reproduce the behavior:
1. Output s2p file with v0.3.3.
2. Open the s2p file with v0.2.2-1.
3. S21 looks like S11.

**Expected behavior**
See S21 data.

**Screenshots**
[image]
Green is the data acquired in v0.3.3,
Dark blue is the expected data obtained with v0.2.2-1.

**Desktop (please complete the following information):**
 - OS: Windows10 1909
 - NanoVNA-Saver version: v0.3.3

KobaSisyo, 2020-07-01:
If you load the s2p file generated by V0.2.2-1 with v0.3.3, s2p is displayed correctly.

zarath, 2020-07-01:
Can you supply me the s2p files? Best one saved with v0.2.2-1 and one saved with v0.3.3

ikd-sci, 2020-07-01:
Second that. Version 0.33. 1. Load data from the examples: Files > Load as sweep > "test/data/attenuator-0643_DB.s2p"
2. Save the data as s2p: Files > Save 2-port file (S2p) > "test.s2p" 3. Load it back, S21 looks exactly like S11

Suggested fix:
NanoVNASaver.py, function exportFile, Line 591:
ts.sdata[1] = self.data21
instead of
ts.sdata[1] = self.data

zarath, 2020-07-01:
yes, found out the same.
if able to install from source

https://github.com/zarath/nanovna-saver/tree/development

should work

zarath, 2020-07-03:
Fixed in current Development Tree

KobaSisyo, 2020-07-04:
This bug seems to have been fixed in pre0.3.4(Win10)!

closed 2020-07-04 by zarath