NanoVNA-Saver/nanovna-saver #224: 201 datapoints setting is not possible.

PA3CCE, 2020-07-03:
I'm runnig NanoVNA-Saver on Windows 10 using NanoVNA-H4 with latest firmware by DiSlord.

Steps to reproduce the behavior:
1. Go to Main Window
2. Click on "Manage" button
3 Goto "Settings/Datapoints" in the "Device settings" window
4. See error :  201 points setting is not selectable.

**Expected behavior**
Setting 201 points should be selectable in a pulldown menu.

zarath, 2020-07-04:
Is H4 otherwise working with v0.3.4 and latestet DiSlord firmware?

I've seen where the problem is burried.. fix will follow soon

RandMental, 2020-07-04:
It seems to be working now. 

Only spesific item on H and H4 is the Manage - > Settings that don't offer 201 data points. I could not find last night where the 201, 101 list gets lost.

PA3CCE, 2020-07-06:
As far as I know yes, I'm still trying out.
Jos

closed 2020-07-06 by PA3CCE