NanoVNA-Saver/nanovna-saver #230: Marker settings

PA3CCE, 2020-07-06:
**Describe the bug**
Strange behaviour after having fllled the markers tn the main window of NanoVNA-Saver 3.4..

**To Reproduce**
Steps to reproduce the behavior:

1. Connect your 2-port device , switchon NanoVNA and  startup NanoVNA-Saver.

2. Click on ''Connect to Device"
   Wait unti;l the sweep has finished.
   The marker data- AND start/stop  area in the main window have been filled with erratic data.

3. Fill in the sweep range I used  50M to 900M
   Fill the marker frequency 1 to 6  bars,  I used 50M, 100M 200M 300M 500M, 900M

4. Start the  sweep,  I had set for 20 Segments in NanoVNMA-Saver.

5. The sweep starts and fills all marker areas wiith data of MARKER 1

**Expected behavior**
Marker areas have to be filled wit data. of markers being set

zarath, 2020-07-06:
Confirmed

zarath, 2020-07-06:
Seems it only occurs in multi segment sweep

PA3CCE, 2020-07-06:
OK, I have seen many more strange data !

zarath, 2020-07-08:
Why do you have closed this bug?
It'a problem, that the position of the marker is invalid, as in segmented scan only the first of the segments is valid. After scan of that segment, it updates already some data, which triggers the markers to update.
I have to investigate a clean solution for that.

PA3CCE, 2020-07-09:
Sorry, I closed this unintentially.

zarath, 2020-07-09:
@PA3CCE 
can you try my latest development update https://github.com/zarath/nanovna-saver/tree/development

PA3CCE, 2020-07-09:
Yes I would like to, can you please send me the executable to my home address jrs@xs4all.nl, I'm not familiar to compiling in Python, sorry I should..
Jos

zarath, 2020-07-09:
I'll tell you when I've made new binaries

zarath, 2020-07-10:
https://github.com/NanoVNA-Saver/nanovna-saver/releases/tag/v0.3.5

PA3CCE, 2020-07-10:
Hi Zarath,
I downloaded and installed the new version 3.5, but sorry I can't get it connected Istarts up but when I click "Connect the device" the program aborts, the COM port is shown correctly (COM11). Version 3.4  still connects flawslesly.

zarath, 2020-07-10:
Can you look at the debug output?
Nanovna-saver.exe -D debug.txt

PA3CCE, 2020-07-10:
Commandline window says : NanoVNASaver.Hardware-Warning-Did not regognize NanVNA type from firmware.
NsanoVNA-Saver window says : "Unable to parse frequency from input start and stop fields".

PA3CCE, 2020-07-10:
This is the complete debug .txt
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4903859/debug.txt)

zarath, 2020-07-10:
I have uploaded fresh binaries. If this still fails. Please post the debug.txt again.

PA3CCE, 2020-07-10:
OK

PA3CCE, 2020-07-10:
I downloaded the executable from the same link and executed it. The program loaded with the same error "could not parse Start and Stop frequency".
I saw that the start frequency was 50k and the stop freq 50.2k so a span of 200 Hz, Segments was set to 20, since the program on a H4 sets to ro 201 points it calculates correctly a step of (200/20)/201 = 49.75 mHz, imho THIS IS THE PROBLEM. !!
I attached the new debug.txt
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4904199/debug.txt)

PA3CCE, 2020-07-10:
I downloaded the file from the old link (above) did I realy get the updated version ? the warning in the command window is still the same also ! The program still aborts when I fill in more reasonable frequencies.

zarath, 2020-07-10:
if it detects -H4 as NanoVNA it is the right one..
Technically it is impossible to run steps below 1 Hz, as Frequencies are transfered as integers.
Can you try Span of 100M or so?

PA3CCE, 2020-07-10:
A setting for mHz per step is impossaible ofcourse, so this must be prohibited I think.
Before dinner I had set for a sweep 50MHz to 2GHz, when I pressed "Sweep"the program aborted.
When I came back the program starts up normally, no error messages, when I press "Connect to device" the program aborts after 1 or 2 seconds as before.

PA3CCE, 2020-07-10:
The last message was when I started NanoVNA-Saver normally.
Now I started NanoVNA-Saver from the command prompt and got the errormessage from NanoVNA-Saver "Unable to parse frequencies" again,  no error- or warning messages in the command window..... Then when I set for 50M to 2G sweep the program aborts after pressing sweep.

zarath, 2020-07-10:
Is the -H4 really capable of 2GHz?

PA3CCE, 2020-07-10:
Yes can go upto 2.7GHz

PA3CCE, 2020-07-10:
Much more noise ofcourse, but it accepts.

PA3CCE, 2020-07-10:
On the high end you'd better use 10 Hz bandwidth and averaging, taking much time. But can be done. (less accurate ofcouse).

igor-m, 2020-07-11:
There is a difference v034 vs. v035 - it seems there is a bug in v035vcalculation somewhere. Here is the v034 and v035 with the same settings and connected to an H-4 terminated with stock coax and stock 50ohm load.
[image]
[image]

PA3CCE, 2020-07-11:
Hi igor-m,
How did you manage to run NanoVNA-Saver v 3.5 ? With me it aborts when pressing  "Connect to device".
My system Windows 10.

PA3CCE, 2020-07-11:
Igor-m,
I think you are not using NanoVNA-H4.

zarath, 2020-07-11:
@PA3CCE i have changed the binaries again. At least the crash should now be recorded in debug log
so nanovna-saver.exe -D debug.txt should give more information why ...

igor-m, 2020-07-11:
@PA3CCE - indeed, I've been using H4. Now freshly flashed with DiSlord's 0847fw (works great on the first glance, the latest fw release before the 09xxfw "SDcard" ones). The saver v035 was broken somehow, now the v036pre seems to work better. Zarath does a great job!

PA3CCE, 2020-07-11:
OK Zarath and and igor-m, I will have dinner first and then will download and run the new upload..... igor-m I'm using latest firmware from DiSlord V0.9.3.1.

PA3CCE, 2020-07-11:
I downloaded and ran the new uploaded version 3.5. The program still aborts when I press "Connect to device". Attached the new debug.txt.
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4907391/debug.txt)

PA3CCE, 2020-07-11:
The problem appears to be in statement exec_command(data 0),  after reading VNA data 0.

igor-m, 2020-07-11:
@PA3CCE - the 09xx fw is for H4 with SDcard mounted, afaik. Also you want the "v036-pre" to get better results, as I wrote above..

PA3CCE, 2020-07-11:
@ igor-m......Yes I have installed a SDcard slot and a clock X-tal and it works very well stand-alone on my NanoVNA-H4.....I just found that there is a version 0.9.3.4 firmware by DiSlord,......I missed that....I just downloaded it and will flash it,  will see what happens....I think it has minor updates.....On the link Zarath gave me I do not find version 0.3.6-pre of NanoVNA-Saver.

PA3CCE, 2020-07-11:
Just flashed the latest firmware 0.9.3.4 by DiSlord and ran NanoVNA-Saver from command prompt, same problem NanoVNA-Saver latest version 3.5 aborts after "Connect to device" .
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4907660/debug.txt)

RandMental, 2020-07-13:
I ran the latest NanoVNA-Saver last night, also with 0.9.3.4 firmware on the H4, had no problems with the USB  connecting and using Saver. 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/NanoVNA-Saver/nanovna-saver/issues/230#issuecomment-657136602",
"url": "https://github.com/NanoVNA-Saver/nanovna-saver/issues/230#issuecomment-657136602",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

zarath, 2020-07-13:
@RandMental was it the v0.3.6-pre version from last midnight?

If yes, I've made a binary here, so @PA3CCE can check
https://github.com/NanoVNA-Saver/nanovna-saver/releases/tag/v0.3.6

RandMental, 2020-07-13:
Hi Holger 

 

Attached is my Debug.txt – I do not have any problems with either 0.3.5 or 0.3.6 Pre to connect and function as required.  Could not duplicate the retries seen by PAC

 

Here is the 0.3.6 debug file – only problem here is that it does not read the  Start and End frequencies from the device on start-up, I guess you have hard-code it to 27 and 30MHz  

 

https://github.com/NanoVNA-Saver/nanovna-saver/releases/tag/v0.3.6

2020-07-13 10:49:10,787 - NanoVNASaver - INFO - Startup...
2020-07-13 10:49:10,849 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-13 10:49:10,849 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-13 10:49:10,992 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-13 10:49:11,197 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.4"
2020-07-13 10:49:11,197 - NanoVNASaver.Windows.About - INFO - This is 0.3.6-pre
2020-07-13 10:49:11,249 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-13 10:49:11,716 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-13 10:49:11,719 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-13 10:49:15,659 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-13 10:49:15,762 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-13 10:49:15,915 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-13 10:49:15,915 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-13 10:49:16,071 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 10:49:16,071 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-13 10:49:16,072 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-13 10:49:16,250 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 10:49:16,251 - NanoVNASaver.Hardware.VNA - DEBUG - result:
Commands: version reset freq offset bandwidth time dac saveconfig clearconfig data frequencies port stat gain power sample scan sweep test touchcal touchtest pause resume cal save recall trace marker edelay capture vbat vbat_offset transform threshold help info color i2c lcd
2020-07-13 10:49:16,453 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 10:49:16,642 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 10:49:16,642 - NanoVNASaver.Hardware.VNA - DEBUG - result:
Board: NanoVNA-H 4
2016-2020 Copyright @edy555
Licensed under GPL. See: https://github.com/ttrftech/NanoVNA
Version: 0.9.3.4 beta FAT16/FAT32/ExFAT- SD Card, by DiSlord
Build Time: Jun 20 2020 - 16:38:20
Kernel: 4.0.0
Compiler: GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Architecture: ARMv7E-M Core Variant: Cortex-M4F
Port Info: Advanced kernel mode
Platform: STM32F303xC Analog & DSP
2020-07-13 10:49:16,643 - NanoVNASaver.Hardware.Hardware - INFO - Type: NanoVNA-H4
2020-07-13 10:49:16,643 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-13 10:49:16,799 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 10:49:16,799 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-13 10:49:16,799 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-13 10:49:16,799 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 10:49:16,959 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 10:49:16,959 - NanoVNASaver.Hardware.VNA - DEBUG - result:
Board: NanoVNA-H 4
2016-2020 Copyright @edy555
Licensed under GPL. See: https://github.com/ttrftech/NanoVNA
Version: 0.9.3.4 beta FAT16/FAT32/ExFAT- SD Card, by DiSlord
Build Time: Jun 20 2020 - 16:38:20
Kernel: 4.0.0
Compiler: GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Architecture: ARMv7E-M Core Variant: Cortex-M4F
Port Info: Advanced kernel mode
Platform: STM32F303xC Analog & DSP
2020-07-13 10:49:16,959 - NanoVNASaver.Hardware.NanoVNA_H4 - INFO - VNA has 201 datapoints capability
2020-07-13 10:49:16,959 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-13 10:49:16,959 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 10:49:17,121 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
[... 35 lines cut ...]
2020-07-13 10:49:18,443 - NanoVNASaver.SweepWorker - DEBUG - Sending "updated" signal
2020-07-13 10:49:18,443 - NanoVNASaver.SweepWorker - DEBUG - Sweep segment no 1
2020-07-13 10:49:18,446 - NanoVNASaver.SweepWorker - INFO - Reading from 28503681 to 29999881. Averaging 1 values
2020-07-13 10:49:18,446 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
2020-07-13 10:49:18,446 - NanoVNASaver.SweepWorker - DEBUG - Setting sweep range to 28503681 to 29999881
2020-07-13 10:49:18,446 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readFrequencies: scan_mask
2020-07-13 10:49:18,446 - NanoVNASaver.SweepWorker - DEBUG - Reading data 0
2020-07-13 10:49:18,447 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readValue with scan mask (data 0)
2020-07-13 10:49:18,447 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(scan 28503681 29999881 201 0b110)
2020-07-13 10:49:19,529 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 7
2020-07-13 10:49:19,529 - NanoVNASaver.SweepWorker - DEBUG - Read 201 values
2020-07-13 10:49:19,530 - NanoVNASaver.SweepWorker - DEBUG - Reading data 1
[... cut ...]
2020-07-13 10:49:19,566 - NanoVNASaver.SweepWorker - DEBUG - Resetting NanoVNA sweep to full range: 27000000 to 30000000
2020-07-13 10:49:19,566 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(sweep 27000000 30000000 201)
2020-07-13 10:49:19,721 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 10:49:19,721 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(resume)
2020-07-13 10:49:19,876 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 10:49:19,876 - NanoVNASaver.SweepWorker - DEBUG - Sending "finished" signal

igor-m, 2020-07-13:
Marker settings issue with latest 036pre - when I set Markers to certain fixed values (ie 1-30, 2-100, 3-300 MHz) and do sweep (10 segments from START to 500MHz) while changing the START frequency (START always below 30MHz, like 2-15MHz) some Markers (usually Marker1, but others too) change its value off expected range (for example 1-4.2MHz, 3-386MHz). Below a debug.txt from such a session, if that helps.
[debug_Markers.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4911911/debug_Markers.txt)

PA3CCE, 2020-07-13:
@Zarath ....I downloaded and installed Nan0VNA-saver V 3.6 pre..... In the mean time I have upgraded to DiSlord's lastest version firmware V 0.9.3.4. ...Sorry,  I still have the same problem,....NanoVNA-Saver loads correctly but whe I press "Connect to device" the sweep starts but after a few seconds the program aborts....., I tried in Windows 10 directly and from a command prompt as well (same result).
Below the debug.txt file

[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4912157/debug.txt)

PA3CCE, 2020-07-13:
@Zarath... After reading the debug file I can see that the problom occurs in the first segment of scanning (36 retry's) and processing the data (to many retry's) of a segmented scan.

igor-m, 2020-07-13:
The crash with latest 036pre happens here as well sometimes. It seems the success depends on the internal state of the H4.  Try to OFF/ON the H4 before you click Connect.

PA3CCE, 2020-07-13:
@Zarath....After reading the debug file of RandMental I think the problem is in the sweeprange/segments setting, I used the default setting and 20 sgments... RandMental had set to a different range and 2 segments only..  Nevertheless RandMental had 10 retry's in reading the first data, but could finish the scan.

zarath, 2020-07-13:
strange, that it tooks so long to get data from your vna...

After scan command it tooks some time until actual data is delivered... My current retry value was 40 (times 50ms = 2seconds)
I've made a 100times retry one (5 seconds)

zarath, 2020-07-13:
@PA3CCE heaving retries is normal, I do repetitve reads to get the data as soon as possible. 10 means, that it only took 0.5seconds after the scan started until data was ready.

igor-m, 2020-07-13:
FYI - enclosed are some logs where it crashed here. 

[debug_ERR1.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4912460/debug_ERR1.txt)
[debug_ERR3.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4912462/debug_ERR3.txt)
[debug_ERR2.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4912463/debug_ERR2.txt)

PA3CCE, 2020-07-13:
I tryed many times to start normally from windows but program aborted always, even when I had set for a different sweeprange before I pressed "Connect to device". @ Zarath Maybe because I had set to 20 segments Nan0VNA-Saver needed more time ?
Just I guess.

PA3CCE, 2020-07-13:
I tried many times to set NanoVNA on/off , always crashes".... now I found it crashes when I press "Sweep Settings" as well, not for the other buttons. I can see that Igor-m's scans run much longer, probably there are more criterea for a crash ?

PA3CCE, 2020-07-13:
I started NanoVNA-Saver from the command prompt, pressed "Sweep Settings",  the command window showed ..................................              Traceback ((most recent call last):
  File "NanoVNASaver\widgets\SweepControl.py", line 92, in <lambda>
  File "NanoVNAsAver\NanoVNASaver.py", line 779, in display_window
KeyError: ''settings

Attached Debug2.txt
[debug2.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4912663/debug2.txt)

RandMental, 2020-07-13:
I have now increased the segments also to 20 and then back to 10.

All works fine on the -H4 0.9.3.4 LSE

 

It now don’t crash in Calibration anymore,  but when I tried to Save the calibration data (10 segments) 

 

regards

 

2020-07-13 15:46:54,908 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-13 15:47:04,472 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-13 15:47:04,576 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-13 15:47:04,730 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-13 15:47:04,730 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-13 15:47:04,884 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:47:04,884 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-13 15:47:04,885 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-13 15:47:05,045 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:47:05,045 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-13 15:47:05,246 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 15:47:05,420 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:47:05,420 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-13 15:47:05,420 - NanoVNASaver.Hardware.Hardware - INFO - Type: NanoVNA-H4
2020-07-13 15:47:05,420 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-13 15:47:05,575 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:47:05,575 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-13 15:47:05,575 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-13 15:47:05,575 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 15:47:05,740 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:47:05,740 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-13 15:47:05,740 - NanoVNASaver.Hardware.NanoVNA_H4 - INFO - VNA has 201 datapoints capability
2020-07-13 15:47:05,740 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-13 15:47:05,740 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 15:47:05,905 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:47:05,905 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-13 15:47:05,905 - NanoVNASaver.Hardware.NanoVNA_H4 - INFO - VNA has 201 datapoints capability
2020-07-13 15:47:05,905 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readFrequencies: scan_mask
2020-07-13 15:47:05,905 - NanoVNASaver.NanoVNASaver - INFO - Read starting frequency 850000000 and end frequency 925000000
2020-07-13 15:47:05,906 - NanoVNASaver.NanoVNASaver - DEBUG - Starting initial sweep
2020-07-13 15:47:05,906 - NanoVNASaver.NanoVNASaver - DEBUG - Starting worker thread
2020-07-13 15:47:05,907 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-13 15:47:05,909 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 402
2020-07-13 15:47:05,909 - NanoVNASaver.SweepWorker - DEBUG - Sweep segment no 0
2020-07-13 15:47:05,909 - NanoVNASaver.SweepWorker - INFO - Reading from 850000000 to 887406400. Averaging 1 values
2020-07-13 15:47:05,909 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
2020-07-13 15:47:05,909 - NanoVNASaver.SweepWorker - DEBUG - Setting sweep range to 850000000 to 887406400
2020-07-13 15:47:05,909 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readFrequencies: scan_mask
2020-07-13 15:47:05,910 - NanoVNASaver.SweepWorker - DEBUG - Reading data 0
2020-07-13 15:47:05,910 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readValue with scan mask (data 0)
2020-07-13 15:47:05,910 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(scan 850000000 887406400 201 0b110)
2020-07-13 15:47:07,256 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 10
2020-07-13 15:47:07,256 - NanoVNASaver.SweepWorker - DEBUG - Read 201 values
2020-07-13 15:47:07,256 - NanoVNASaver.SweepWorker - DEBUG - Reading data 1
2020-07-13 15:47:07,256 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readValue with scan mask (data 1)
2020-07-13 15:47:07,256 - NanoVNASaver.SweepWorker - DEBUG - Read 201 values
2020-07-13 15:47:07,257 - NanoVNASaver.SweepWorker - DEBUG - Averaging 1 values
2020-07-13 15:47:07,257 - NanoVNASaver.SweepWorker - DEBUG - Calculating data and inserting in existing data at index 0
2020-07-13 15:47:07,258 - NanoVNASaver.SweepWorker - DEBUG - update Freqs: 201, Offset: 0
2020-07-13 15:47:07,258 - NanoVNASaver.SweepWorker - DEBUG - Saving data to application (402 and 402 points)
2020-07-13 15:47:07,258 - NanoVNASaver.SweepWorker - DEBUG - Sending "updated" signal
2020-07-13 15:47:07,258 - NanoVNASaver.SweepWorker - DEBUG - Sweep segment no 1
2020-07-13 15:47:07,258 - NanoVNASaver.SweepWorker - INFO - Reading from 887593432 to 924999832. Averaging 1 values
2020-07-13 15:47:07,258 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
[... 602 lines cut ...]
2020-07-13 15:49:02,159 - NanoVNASaver.SweepWorker - DEBUG - Sweep segment no 1
2020-07-13 15:49:02,159 - NanoVNASaver.SweepWorker - INFO - Reading from 857503732 to 864970132. Averaging 1 values
2020-07-13 15:49:02,159 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
2020-07-13 15:49:02,159 - NanoVNASaver.SweepWorker - DEBUG - Setting sweep range to 857503732 to 864970132
[... cut ...]
2020-07-13 15:49:03,415 - NanoVNASaver.SweepWorker - INFO - Reading from 865007464 to 872473864. Averaging 1 values
2020-07-13 15:49:03,415 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
2020-07-13 15:49:03,415 - NanoVNASaver.SweepWorker - DEBUG - Setting sweep range to 865007464 to 872473864
2020-07-13 15:49:03,415 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readFrequencies: scan_mask
2020-07-13 15:49:03,415 - NanoVNASaver.SweepWorker - DEBUG - Reading data 0
2020-07-13 15:49:03,415

igor-m, 2020-07-13:
@RandMental - a hint: you may attach your debug.txt file into your message by dragging and dropping..

PA3CCE, 2020-07-13:
@ RandMental ....Very strange, I'm running NanoVNA-H4 on DiSlord's version 0.9.3.4 on NanoVNA-Saver 3.6 pre as well and can't get it connected, ....It crahes while doing the initial sweep on the default settings....Switching the device on and off before does not help.

zarath, 2020-07-13:
> FYI - enclosed are some logs where it crashed here (latest 036pre, H4-0847fw, Win10, 10Segments).
> 
> [debug_ERR1.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4912460/debug_ERR1.txt)

Validation Error.

Sorry I have no clue why Rune had added an exception if a data value is out of an arbitrarily range.
The value was raised several times.. I don't know if there was a discussion about set setting in the early state of the software...

RandMental, 2020-07-13:
Correction - calibration still fails with a crash when clicking Apply  after the Wizard finished the SOL calibration sequence.   10 Segments by 201 points sweep selected

 

Debug attached

 

regards

 

 

 

 

 

 

2020-07-13 15:55:00,890 - NanoVNASaver - INFO - Startup...
2020-07-13 15:55:00,953 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-13 15:55:00,953 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-13 15:55:01,102 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-13 15:55:01,661 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.4"
2020-07-13 15:55:01,661 - NanoVNASaver.Windows.About - INFO - This is 0.3.6-pre
2020-07-13 15:55:01,706 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-13 15:55:02,169 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-13 15:55:02,176 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-13 15:55:05,627 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-13 15:55:05,730 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-13 15:55:05,885 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-13 15:55:05,885 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-13 15:55:06,040 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:55:06,040 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-13 15:55:06,040 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-13 15:55:06,199 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:55:06,199 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-13 15:55:06,400 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 15:55:06,582 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:55:06,582 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-13 15:55:06,583 - NanoVNASaver.Hardware.Hardware - INFO - Type: NanoVNA-H4
2020-07-13 15:55:06,583 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-13 15:55:06,740 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:55:06,740 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-13 15:55:06,740 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-13 15:55:06,740 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 15:55:06,901 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:55:06,901 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-13 15:55:06,901 - NanoVNASaver.Hardware.NanoVNA_H4 - INFO - VNA has 201 datapoints capability
2020-07-13 15:55:06,901 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-13 15:55:06,902 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-13 15:55:07,063 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:55:07,063 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-13 15:55:07,063 - NanoVNASaver.Hardware.NanoVNA_H4 - INFO - VNA has 201 datapoints capability
2020-07-13 15:55:07,063 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readFrequencies: scan_mask
2020-07-13 15:55:07,063 - NanoVNASaver.NanoVNASaver - INFO - Read starting frequency 850000000 and end frequency 925000000
2020-07-13 15:55:07,064 - NanoVNASaver.NanoVNASaver - DEBUG - Starting initial sweep
2020-07-13 15:55:07,064 - NanoVNASaver.NanoVNASaver - DEBUG - Starting worker thread
2020-07-13 15:55:07,065 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-13 15:55:07,068 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 2010
2020-07-13 15:55:07,068 - NanoVNASaver.SweepWorker - DEBUG - Sweep segment no 0
2020-07-13 15:55:07,069 - NanoVNASaver.SweepWorker - INFO - Reading from 850000000 to 857466400. Averaging 1 values
2020-07-13 15:55:07,069 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
2020-07-13 15:55:07,069 - NanoVNASaver.SweepWorker - DEBUG - Setting sweep range to 850000000 to 857466400
2020-07-13 15:55:07,069 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readFrequencies: scan_mask
[... 591 lines cut ...]
2020-07-13 15:57:16,736 - NanoVNASaver.SweepWorker - DEBUG - Averaging 1 values
2020-07-13 15:57:16,742 - NanoVNASaver.SweepWorker - DEBUG - Calculating data and inserting in existing data at index 9
2020-07-13 15:57:16,742 - NanoVNASaver.SweepWorker - DEBUG - update Freqs: 201, Offset: 1809
2020-07-13 15:57:16,743 - NanoVNASaver.SweepWorker - DEBUG - Saving data to application (2010 and 2010 points)
2020-07-13 15:57:16,912 - NanoVNASaver.SweepWorker - DEBUG - Sending "updated" signal
2020-07-13 15:57:16,912 - NanoVNASaver.SweepWorker - DEBUG - Resetting NanoVNA sweep to full range: 850000000 to 925000000
2020-07-13 15:57:16,912 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(sweep 850000000 925000000 201)
2020-07-13 15:57:17,107 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:57:17,107 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(resume)
2020-07-13 15:57:17,261 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-13 15:57:17,261 - NanoVNASaver.SweepWorker - DEBUG - Sending "finished" signal
2020-07-13 15:57:32,426 - NanoVNASaver.NanoVNASaver - DEBUG - Starting worker thread
2020-07-13 15:57:32,428 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
[... cut ...]
2020-07-13 15:57:32,429 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readValue with scan mask (data 0)
2020-07-13 15:57:32,429 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(scan 850000000 857466400 201 0b110)
2020-07-13 15:57:33,959 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 12
2020-07-13 15:57:33,959 - NanoVNASaver.SweepWorker - DEBUG - Read 201 values
2020-07-13 15:57:33,959 - NanoVNASaver.SweepWorker - DEBUG - Reading data 1
2020-07-13 15:57:33,959 - NanoVNASaver.Hardware.NanoVNA

PA3CCE, 2020-07-13:
NanoVNA-Saver 3.6 pre freezes when started from scratch
=====================================

closed 2020-07-15 by zarath