NanoVNA-Saver/nanovna-saver #233: Erratic display behaviour with "Hide Data" in Markers

igor-m, 2020-07-09:
When clicking on "Hide Data" in Markers section the display starts to jump erratically - the graphs (ie 3 selected) several pixels left-right direction and the "Reference sweep" part on the left several pixels up/down direction.
No erratic jumps when the marker data are shown.
It does with or without nanovna-h4 connected.
Nanovna-Saver 0.3.4
Win 10 64bits

igor-m, 2020-07-10:
The picture jerks when ie.:
a) the main app window is maximized
a) there are 3 graphs selected
b) one of them is Polar Plot or Smith Chart
[image]

RandMental, 2020-07-10:
I can not duplicate this 

 

 <https://user-images.githubusercontent.com/7240123/87146164-d0b56880-c2aa-11ea-908a-5a7fa4e2a7a2.PNG>

igor-m, 2020-07-10:
Here an example in .avi
[Erratic display.zip](https://github.com/NanoVNA-Saver/nanovna-saver/files/4902709/Erratic.display.zip)

closed 2022-11-10 by github-actions[bot]