NanoVNA-Saver/nanovna-saver #237: Note: running nanovna-saver from Linux docker image

grahamwhaley, 2020-07-10:
**Is your feature request related to a problem? Please describe.**

This is not an issue of the application - but I had great problems trying to get Python3.7 installed and running on my Ubuntu 18.04 system as a native setup. I resorted to running nanovna-saver in a docker image, with the associated setup tweaks to allow access to the host screen and USB device.

I placed a few details in a [gist](https://gist.github.com/grahamwhaley/907dce0957036f16cfa7513160d261ec), and also pushed the pretty basic files up to a [branch in my fork](https://github.com/grahamwhaley/nanovna-saver/tree/20200710_run_docker).

**Describe the solution you'd like**

If you feel running out of a docker image/container is a viable install/run method that merits documenting in the main repo, then I'm happy to work with somebody on how/where to document this etc., and send a PR.

**Describe alternatives you've considered**

I tried a Python virtual env before I tried docker. But, afaict, it looks like python3.7 is not exactly mainline supported on Ubuntu 18.04? I also considered trying another linux distro, but did not have one trivially to hand right now.

**Additional context**

There may be wrinkles to do with save/load of settings or defaults, and maybe the mounted volume need discussing. I just hacked it together enough to make it work. I'm very new to nanovna-saver and the nanovna, so I may have missed some obvious things.

There are a few warnings issued from the docker container about missing drivers (such as for i915 on my system) - so it may be possible to improve the `Dockerfile` to add more drivers that may accelerate QT5 - just a guess.

And, the name of the USB tty device is probably not guaranteed, depending on the distro and end users system. Maybe that could be automated in a script to locate the correct device to pass through.

zarath, 2020-07-10:
In docker you'll get fun with X and the serial devices...
Have you tried the binary "ubuntu" distribution? I'ts not so ubuntu specific and bundles a bunch of .so files..

grahamwhaley, 2020-07-10:
Doh @zarath - I missed there were binary releases! - I'll go spin that. Thanks!
I got the docker X and USB working - but, sure, depending on systems with USB device naming and different compositors (wayland etc.), it would either be prone to error or complex to make fairly generic.

grahamwhaley, 2020-07-10:
Thank you @zarath - the binary release looks to work well.
I'll close this Issue - but, am glad to have it here in the github history so maybe somebody will find the information or solution useful in the future.

closed 2020-07-10 by grahamwhaley