NanoVNA-Saver/nanovna-saver #238: Calibration buttons do not provide cal data

igor-m, 2020-07-11:
While calibrating and clicking a the "Short" "Load" "Open" buttons the action does not provide calibration procedure and data. Calibration procedure works via "Calibration assistant" only.
saver v0.3.4

zarath, 2020-07-14:
should be fixed in v0.3.6-rc

igor-m, 2020-07-15:
Nope, it does nothing here in 036rc2..

zarath, 2020-07-15:
have you pushed the buttons hard enough? ;-)

PA3CCE, 2020-07-16:
@ Zarath.... Hi Zarath did you delete my issues abt aborting NanoVNA-Saver at start from scrath and abt nthe Markers ? This morning I still found some issues.

PA3CCE, 2020-07-16:
@Zarath.... I downloaded and installed rc2, ran it and it performed vy well, so this is the reason why you closed my previous issues I suppose..... At first glance I saw that, though I did not set any markers, The markers are still filled with data of the Start frequency, and that my **scrolling** the **mousewheel** problem was'nt solved..... I keep on testing. ...regards

PA3CCE, 2020-07-16:
I forgot to attach the debug.txt file.
[test-rc2.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4931764/test-rc2.txt)

zarath, 2020-07-16:
> @zarath.... I downloaded and installed rc2, ran it and it performed vy well, so this is the reason why you closed my previous issues I suppose..... At first glance I saw that, though I did not set [...]

Please make a new bug report, I cannot track a thounsend different bugs in one thread....

zarath, 2020-07-16:
Cannot reproduce, see here:

https://youtu.be/G3kqqY8eDIU

igor-m, 2020-07-16:
After SOL I get error when applying calibration
[image]

zarath, 2020-07-16:
Pushing the short load.. button takes the last runned sweep as  data. so as in my video you have to connect short, make a single sweep and than press short button and so on. Point count is shown immediately

igor-m, 2020-07-16:
Well, then this app is not intended for users with my way of thinking.. :) 
I expected 
1. I connect short
2. push the Short button
3. it makes a sweep
4. it writes "Set 2010 points"
and so on..

closed 2020-07-16 by igor-m

zarath, 2020-07-16:
That's why this is the experts option. Kidding. A user handbook would be nice..