NanoVNA-Saver/nanovna-saver #239: Broken data in graphs (v035)

igor-m, 2020-07-11:
It seems there is a bug in v035 graph calculations or elsewhere. Here is an example with the v034 and v035 - the same settings and connected to an H4 terminated with stock coax and stock 50ohm load (continual sweep, not calibrated).
[image]
[image]

RandMental, 2020-07-11:
Have you Recalibrated the 0.3.5 version?     The jaggered lines normally means you need to recalibrate.  . 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/NanoVNA-Saver/nanovna-saver/issues/239",
"url": "https://github.com/NanoVNA-Saver/nanovna-saver/issues/239",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

igor-m, 2020-07-11:
The graphs are broken when "calibrated" as well - see below. The calibration has been done within saver v035.
[image]

igor-m, 2020-07-11:
The same cal procedure with v034 and sweep
[image]
. Calibration has been provided within v034.

zarath, 2020-07-11:
Very strange behaviour...
Mostly jumps at the 300/600/900/1200 MHz marks

igor-m, 2020-07-11:
All above done with ORIGINAL FACTORY H4 firmware.
I'm going to flash the NanoVNA-H4 v0.8.4.7.dfu in..

zarath, 2020-07-11:
@igor-m are you able to run source code installation?
It seems, that it is only happen with older sweep / scan command.
If I do a scan mask command it's looks allright.

igor-m, 2020-07-11:
The 0847dfu works unreliably with the v305 however, it loads nonsense upon connect into start/stop and usually crashes the app. I was lucky to get the above sweep, it seems..

igor-m, 2020-07-11:
Enclosed the debug.txt (H4-0847fw and v035 saver).
After connect it reads nonsense into STOP frequency and crashes after couple of seconds while 
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4906887/debug.txt)
doing nothing..

zarath, 2020-07-11:
ok. crash is because of data validation... there are very high data readings. And there is a code path that then simply crashes :-(
So if data validation is disabled it should run further ...

igor-m, 2020-07-11:
This is a "successful" start, params as above, but I OFF/ONed the H4, then started v035, the START/STOP was empty, then I connected H4, it read the H4 default START/STOP, the sweep has been done ok, but graphs with glitches (see below). Attached the debug.txt from this session..
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4906902/debug.txt)
[image]

igor-m, 2020-07-11:
This is as above but with Calibration (50kHz/900Mhz). Still glitches in the graphs.

[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4906929/debug.txt)
[image]

zarath, 2020-07-11:
can you try https://github.com/zarath/nanovna-saver/actions/runs/165679524

igor-m, 2020-07-11:
After connecting freshly started H4 (default 50k/900M, internally calibrated) it put start/stop 27/30MHz and did a sweep.
I entered 50k/900M and did a sweep - see below (2010points). Looks promising..
[image]

zarath, 2020-07-11:
So I just have to investigate which firmwares are able to do scan mask command >= 0.7.1 of original nanovna...

closed 2022-11-10 by github-actions[bot]