NanoVNA-Saver/nanovna-saver #242: does not run at WIN10 32 bit.

dg3seg, 2020-07-13:
**Describe the bug**
Is it a bug or feature? When I try to start the nanovna-saver.exe (0.3.5) only the message that this version is not compatible with my Windows 10 appears. The start is not possible.

Would you please be so kind to give me a solution for this?
Maybe I do something wrong but I guess this is only for Win 10 64bit runnable.
Kind regards
DG3SEG
----------- This is the output--------------
nanovna-saver.exe ist mit der ausgeführten Windows-Version nicht kompatibel. Überprüfen Sie die Systeminformationen des Computers, und wenden Sie sich anschließend an den Herausgeber der Software.

zarath, 2020-07-13:
If you can supply a 32bit build environment on github, please feel free to do.

First line about binaries in README.md tells that currently 64bit only.

zarath, 2020-07-14:
of course source installation should work on win32

closed 2020-07-14 by zarath