NanoVNA-Saver/nanovna-saver #244: NanoVNA-Saver 3.6 pre freezes when started from scratch from scratch

PA3CCE, 2020-07-13:
**Describe the bug**
A clear and concise description of what the bug is.

When You install NanoVNA_saver 3.6 pre from scratch, the program starts-up,...When you click on " Connect to device" the program freezes.

**To Reproduce**
Steps to reproduce the behavior:

1.  When you had NanoVNA_Saver earlier installed, delete the program.
     Goto C:/Users/yourname/appdata/Roaming and delete repository "NanoVNA_Saver"
2.   Download NanoVNA-Saver 3.6 pre and install.
3.   Run NanoVNA-Saver and answer question about update.
4.   Press "Manage" button and deactivate the default "Validate received data"  (Seems to create an abort).
5.  Press "Connect to device".
6. The program freezes. (Important for new users !!!!)

**Expected behavior**
The program connects to the device and is ready to be used.

**Desktop (please complete the following information):**
 - OS:  Windows 10 Home edition (fully updated).
 - NanoVNA-Saver version: 3.6 pre.
 - NanoVNA-H4 with DiSlord's firmwaware V 3.9.4 SLE (with SSD slot and clock X-tal).

**Additional context**
After a lot of problems I wanted to start from scratch.

igor-m, 2020-07-13:
I've done as per your above steps.
It works here (H4 0847fw).
This is what I see after I've clicked on Sweep (H4 CH0 terminated by the stock load).
[image]

PA3CCE, 2020-07-13:
@ Igor-m....If I understand well, you are using different firmware (0847fw)....Did you try DiSlord's latest firmware ? it's very good !!!, Different firmware might cause different results.

igor-m, 2020-07-13:
I did with 0847fw (the latest pre-SDcard one), but there is an another issue I cope with, so I downgraded to 0846fw - the same, now I test 0845 firmware. I do not want go to the 09xx as it is much more complex (sdcard etc). I want to see a stable operation first..

PA3CCE, 2020-07-13:
OK I see, well with me,  with my firmware,  NanoVNA-Saver old versions (before 3.5) worked well but without the ectra's in the firmware............. I just started the program from window's command prompt to produce a debug.txt file......To my surprise it connected flawlessly and the screen was as yours. ....When I looked for a debug.txt file, there was'nt one .........Because the program started and connected correctly from the prompt, I tried to start and connect the program normally (not from the prompt) and it connected correctly as well....Then I set for a scan of 50M to 900M  and 20 segments and loaded a corresponding calibration file.... This caused an abort of the program (I had a BiasTee connected between CH0 and CH1)....I went back to starting the program from the prompt and now the program aborted after pressing "Connect to the device".... This time I found the debug.text file as attached.................I'm getting frustrated. 
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4915069/debug.txt)

igor-m, 2020-07-13:
Do not be frustrated.. Everything is beta here :) At the end of the day it will work fine, indeed, as the talented people mess with debugging those betas :)

zarath, 2020-07-13:
Some of your crashes should be fixed in latest source code. It is time consuming for me to create a binary on every patch.  As the process of building isn't fully automatic yet

igor-m, 2020-07-14:
You may test without binaries in Linux, for example. Here is Ubuntu in VirtualBox (Win10) running 036-pre Development branch..
[image]

PA3CCE, 2020-07-14:
OK, I will wait for new binaries.

PA3CCE, 2020-07-14:
@ igor-m Yes I know very well this is beta and for the frustation : I was joking.

zarath, 2020-07-14:
@PA3CCE 

fresh binaries! .. Some of your issues could be due to bandwidth settings of nanovna. If your device is in bandwidth 10 mode, it took exponential longer to scan the data, compared to bandwidth 1000 (the default)

As there is no way to read out current setting from the vna, I now set the bandwidth to 1000 on connect.
But now you can select possible values in the device setting dialog.

Hope you have more luck with v0.3.6-rc now

PA3CCE, 2020-07-14:
For your info :  This morning I deleted the NanoVNA-Saver repository from Windows C: drive (The program itself is installed on my D: drive)....Then tried starting the program normally from Windows.... The first time it aborted immediately,... The second time it started up and I could connect, set sweeprange to 50 to 900MHz 1 segment only and run with reasonable result.....Then I tried to load a calibration file  made by NanoVNA-Saver version 3.4,  this caused the program to abort.

zarath, 2020-07-14:
debug.txt?

RandMental, 2020-07-14:
I cannot build 0.3.6 in your dev branch

 

Traceback (most recent call last):

  File "C:\Python\Python38\lib\runpy.py", line 194, in _run_module_as_main

    return _run_code(code, main_globals, None,

  File "C:\Python\Python38\lib\runpy.py", line 87, in _run_code

    exec(code, run_globals)

  File "C:\Workspaces\nanovna-saver-z\NanoVNASaver\__main__.py", line 88, in <module>

    main()

  File "C:\Workspaces\nanovna-saver-z\NanoVNASaver\__main__.py", line 80, in main

    window = NanoVNASaver()

  File "C:\Workspaces\nanovna-saver-z\NanoVNASaver\NanoVNASaver.py", line 87, in __init__

    self.sweep_control = SweepControl(self)

TypeError: 'module' object is not callable

igor-m, 2020-07-14:
While running your 036rc exe

`NanoVNASaver 0.3.6-rc

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Settings: C:/Users/pito/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
Traceback (most recent call last):
  File "nanovna-saver.py", line 24, in <module>
  File "NanoVNASaver\__main__.py", line 80, in main
  File "NanoVNASaver\NanoVNASaver.py", line 87, in __init__
TypeError: 'module' object is not callable
[170552] Failed to execute script nanovna-saver`

igor-m, 2020-07-14:
NanoVNASaver 0.3.6-rc

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Settings: C:/Users/smith/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
Traceback (most recent call last):
  File "nanovna-saver.py", line 24, in <module>
  File "NanoVNASaver\__main__.py", line 80, in main
  File "NanoVNASaver\NanoVNASaver.py", line 87, in __init__
TypeError: 'module' object is not callable
[170552] Failed to execute script nanovna-saver

RandMental, 2020-07-14:
Ditto

 

Something broke ----

 

C:\Python\Python38\python.exe -m NanoVNASaver.__main__ -D debug.txt

NanoVNASaver 0.3.6-rc

 

Copyright (C) 2019, 2020 Rune B. Broberg

Copyright (C) 2020 NanoVNA-Saver Authors

 

This program comes with ABSOLUTELY NO WARRANTY

This program is licensed under the GNU General Public License version 3

 

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

 

Settings: C:/Users/deond/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini

Traceback (most recent call last):

  File "C:\Python\Python38\lib\runpy.py", line 194, in _run_module_as_main

    return _run_code(code, main_globals, None,

  File "C:\Python\Python38\lib\runpy.py", line 87, in _run_code

    exec(code, run_globals)

  File "C:\Workspaces\nanovna-saver-z\NanoVNASaver\__main__.py", line 88, in <module>

    main()

  File "C:\Workspaces\nanovna-saver-z\NanoVNASaver\__main__.py", line 80, in main

    window = NanoVNASaver()

  File "C:\Workspaces\nanovna-saver-z\NanoVNASaver\NanoVNASaver.py", line 87, in __init__

    self.sweep_control = SweepControl(self)

TypeError: 'module' object is not callable

 

Process finished with exit code 1

zarath, 2020-07-14:
Pushed missing __init__.py

RandMental, 2020-07-14:
I get now that Saver crash when connected  -

 

USB2.0 -H

2020-07-14 13:47:26,475 - NanoVNASaver - INFO - Startup...
2020-07-14 13:47:26,547 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-14 13:47:26,548 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-14 13:47:26,887 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-14 13:47:27,101 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.4"
2020-07-14 13:47:27,101 - NanoVNASaver.Windows.About - INFO - This is 0.3.6-rc
2020-07-14 13:47:27,149 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-14 13:47:27,419 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-14 13:47:27,423 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-14 13:47:29,646 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-14 13:47:29,750 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-14 13:47:29,904 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-14 13:47:29,904 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-14 13:47:30,007 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 13:47:30,059 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-14 13:47:30,059 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.7.1-23-g2d3cb35']
2020-07-14 13:47:30,059 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-14 13:47:30,164 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 13:47:30,219 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-14 13:47:30,219 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['Commands:', 'version', 'reset', 'freq', 'offset', 'bandwidth', 'dac', 'saveconfig', 'clearconfig', 'data', 'frequencies', 'port', 'stat', 'gain', 'power', 'sample', 'scan', 'sweep', 'test', 'touchcal', 'touchtest', 'pause', 'resume', 'cal', 'save', 'recall', 'trace', 'marker', 'edelay', 'capture', 'vbat', 'vbat_offset', 'transform', 'threshold', 'help', 'info', 'color']
2020-07-14 13:47:30,219 - NanoVNASaver.Hardware.VNA - DEBUG - get bandwidths
2020-07-14 13:47:30,219 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(bandwidth)
2020-07-14 13:47:30,322 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 13:47:30,372 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0

USB 3.0 -H
2020-07-14 13:48:29,105 - NanoVNASaver - INFO - Startup...
2020-07-14 13:48:29,170 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-14 13:48:29,170 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-14 13:48:29,525 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-14 13:48:30,070 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.4"
2020-07-14 13:48:30,070 - NanoVNASaver.Windows.About - INFO - This is 0.3.6-rc
2020-07-14 13:48:30,117 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-14 13:48:30,395 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-14 13:48:30,399 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-14 13:48:34,055 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-14 13:48:34,159 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-14 13:48:34,311 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-14 13:48:34,311 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-14 13:48:34,415 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 13:48:34,465 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-14 13:48:34,465 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.7.1-23-g2d3cb35']
2020-07-14 13:48:34,465 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-14 13:48:34,570 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 13:48:34,625 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-14 13:48:34,625 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['Commands:', 'version', 'reset', 'freq', 'offset', 'bandwidth', 'dac', 'saveconfig', 'clearconfig', 'data', 'frequencies', 'port', 'stat', 'gain', 'power', 'sample', 'scan', 'sweep', 'test', 'touchcal', 'touchtest', 'pause', 'resume', 'cal', 'save', 'recall', 'trace', 'marker', 'edelay', 'capture', 'vbat', 'vbat_offset', 'transform', 'threshold', 'help', 'info', 'color']
2020-07-14 13:48:34,625 - NanoVNASaver.Hardware.VNA - DEBUG - get bandwidths
2020-07-14 13:48:34,625 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(bandwidth)
2020-07-14 13:48:34,728 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 13:48:34,780 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0

zarath, 2020-07-14:
> For your info : This morning I deleted the NanoVNA-Saver repository from Windows C: drive (The program itself is installed on my D: drive)....Then tried starting the program normally from [...]

strange that v0.3.6-rc binary runs any way.  current should

igor-m, 2020-07-14:
Crashes here as well when trying to connect (running 036-rc from your development branch).

zarath, 2020-07-14:
strange that there is no backtrace .. 
I assume an -H or -H4 version .. 
can anyone use a terminal and issue an naked bandwidth command?

ch>bandwidth

and paste me the result?

igor-m, 2020-07-14:
self.set_bandwidth(self.get_bandwidths()[-1])
IndexError: list index out of range

igor-m, 2020-07-14:
NanoVNA Shell
ch> bandwidth

no result..

zarath, 2020-07-14:
> NanoVNA Shell
> ch> bandwidth
> 
> no result..

daf*q

RandMental, 2020-07-14:
Hi Holger

Both H1 and H4

 

ch> bandwidth                                                                   

bandwidth 1 (1000Hz)                                                            

ch>

igor-m, 2020-07-14:
ch> bandwidth

bandwidth 1 (1000Hz)

(LF/CR problemm :) )

zarath, 2020-07-14:
did i told that i hate dos linebreaks :-)

zarath, 2020-07-14:
plain bandwidth do return something ?
or not?

RandMental, 2020-07-14:
Below Sequence RED is me typing, black is what comes back

 

CRLFCH>  bandwidthCRLFbandwidth 1 (1000Hz)CRLFCH> 

 

Note there is a space adter each “>” character

RandMental, 2020-07-14:
Hallo Holger 

 

The Device now stops sweeping for its own display – seems to hang, but when ones click on Saver -> Sweep, then it does respond.

 

Regards

 

2020-07-14 14:28:34,152 - NanoVNASaver - INFO - Startup...
2020-07-14 14:28:34,223 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-14 14:28:34,223 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-14 14:28:34,571 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-14 14:28:35,108 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.4"
2020-07-14 14:28:35,108 - NanoVNASaver.Windows.About - INFO - This is 0.3.6-rc
2020-07-14 14:28:35,159 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-14 14:28:35,419 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-14 14:28:35,424 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-14 14:28:37,372 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-14 14:28:37,476 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-14 14:28:37,629 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-14 14:28:37,629 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-14 14:28:37,732 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 14:28:37,782 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-14 14:28:37,782 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-14 14:28:37,782 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-14 14:28:37,884 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 14:28:37,939 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-14 14:28:37,939 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['Commands:', 'version', 'reset', 'freq', 'offset', 'bandwidth', 'time', 'dac', 'saveconfig', 'clearconfig', 'data', 'frequencies', 'port', 'stat', 'gain', 'power', 'sample', 'scan', 'sweep', 'test', 'touchcal', 'touchtest', 'pause', 'resume', 'cal', 'save', 'recall', 'trace', 'marker', 'edelay', 'capture', 'vbat', 'vbat_offset', 'transform', 'threshold', 'help', 'info', 'color', 'i2c', 'lcd']
2020-07-14 14:28:37,939 - NanoVNASaver.Hardware.VNA - DEBUG - get bandwidths
2020-07-14 14:28:37,939 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(bandwidth)
2020-07-14 14:28:38,043 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 14:28:38,093 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-14 14:46:42,911 - NanoVNASaver - INFO - Startup...
2020-07-14 14:46:42,978 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-14 14:46:42,979 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-14 14:46:43,331 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-14 14:46:43,855 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.4"
2020-07-14 14:46:43,856 - NanoVNASaver.Windows.About - INFO - This is 0.3.6-rc
2020-07-14 14:46:43,906 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-14 14:46:44,189 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-14 14:46:44,193 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-14 14:55:42,827 - NanoVNASaver - INFO - Startup...
2020-07-14 14:55:42,901 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-14 14:55:42,901 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-14 14:55:43,257 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-14 14:55:43,829 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.4"
2020-07-14 14:55:43,829 - NanoVNASaver.Windows.About - INFO - This is 0.3.6-rc
2020-07-14 14:55:43,876 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-14 14:55:44,156 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-14 14:55:44,159 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-14 14:55:46,235 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-14 14:55:46,337 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-14 14:55:46,490 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-14 14:55:46,600 - NanoVNASaver.Hardware.Hardware - DEBUG - Needed retries: 0
2020-07-14 14:55:46,600 - NanoVNASaver.Hardware.Hardware - INFO - Type: NanoVNA-H4
2020-07-14 14:55:46,600 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-14 14:55:46,703 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 14:55:46,754 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-14 14:55:46,754 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-14 14:55:46,754 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-14 14:55:46,754 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-14 14:55:46,858 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 21.0
2020-07-14 14:55:46,916 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
[... 501 lines cut ...]
2020-07-14 15:04:54,773 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
[... cut ...]
2020-07-14 15:04:56,118 - NanoVNASaver.SweepWorker - DEBUG - Averaging 1 values
2020-07-14 15:04:56,121 - NanoVNASaver.SweepWorker - DEBUG - Calculating data and inserting in existing data at index 0
2020-07-14 15:04:56,121 - NanoVNASaver.SweepWorker - DEBUG - update Freqs: 201, Offset: 0
2020-07-14 15:04:56,121 - NanoVNASaver.SweepWorker - DEBUG - Saving data to application (201 and 201 points)
2020-07-14 15:04:56,121 - NanoVNASaver.SweepWorker - DEBUG - Sending "updated" signal
2020-07-14 15:04:56,122 - NanoVNASaver.SweepWorker - DEBUG - Sending "finished" signal

igor-m, 2020-07-14:
:)
[image]

zarath, 2020-07-14:
> Hallo Holger The Device now stops sweeping for its own display – seems to hang, but when ones click on Saver -> Sweep, then it does respond. Regards 2020-07-14 14:28:34,152 - NanoVNASaver - INFO - Startup... 2020-07-14 14:28:34,223 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker

seems not to be the latest commit, as i now should see the sequence:
2020-07-14 15:15:47,232 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...      
2020-07-14 15:15:47,335 - NanoVNASaver.Hardware.Hardware - DEBUG - Needed retries: 0                                                                                                                            
2020-07-14 15:15:47,335 - NanoVNASaver.Hardware.Hardware - INFO - Type: Generic NanoVNA    
during connect to device

igor-m, 2020-07-14:
I need a bit larger display :)
[image]
[image]

zarath, 2020-07-14:
4k is the way to go :100:

zarath, 2020-07-14:
I don't remember if I had defined a upper limit for the number of markers ...

PA3CCE, 2020-07-14:
@ Zarath........ I downloaded your new version 3.6 rc and installed it in a new repository (deleted the old one) on my D: drive........Deleted the NanoVNA repository from the C: drive....... Started NanoVNA-Saver 3.6 rc from it's repository........ The program started up........ Pressed  "Connect to device"....... The program aborted........ Tried 5 times !!!.,......Then I started the program from Window's command prompt.........The program started up..........Pressed  "Connect to device" ........ The program aborted (tried twice).......,....A repository NanoVNASaver was created on the C: drive  with NanovnaSaver.ini,  no Bands.ini..........Debug.txt is attached
I could not attach the .ini file because here .ini is not supported so I changed it into NanoVNASaver.txt 

[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4919583/debug.txt)
[NanoVNASaver.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4919634/NanoVNASaver.txt)

....... The command-window did not abort and ended with :

Settings : C:/Users/jrs/AppData/Raoming/NanovnaSaver/NanoVNASaver.ini
Traceback (most recent call last) :
  File "NanoVNASaver\NanoVNASaver.py", line 519, in serialButtonClick
  File "NanoVNASaver\NanoVNASaver.py", line 541, in connect_device
  File "NanoVNASaver\Hardware\Hardware.py", line 90, in get_VNA
  File "NanoVNASaver\Hardware\VNA.py", line52, in __init__
Indexerror: List index out of range

igor-m, 2020-07-14:
@PA3CCE - based on your debug file you are trying to run the version 036-pre.
You should run 036rc instead..

PA3CCE, 2020-07-14:
@ igor-m.... Thanks for your warning...., I checked my upload and it is from the 3.6 pre...., I must have picked the file from the wrong repository......I checked the debug file in the correct repository and it shows 3.6 rc, Sorry abt that. I attached the correct file here.

**@** **Zarach...Vy** **sorry** **I** **attached** **the** wrong **debug** file **in** **my** **previous** **message,** **The** **correct** **file** is attached **here.**

**[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4920523/debug.txt)**

igor-m, 2020-07-14:
@PA3CCE - where your repository with 036-rc comes from?

PA3CCE, 2020-07-14:
@igvor-m ....... I don't quite understand why you ask,..... I downloaded the .zip file from Github and extracted in a new repository on my D: drive in a newly created repository called NanoVNA-Sver 3.6 Release Candidate..... That's all.

igor-m, 2020-07-14:
Where from Github did you download the zip file?

PA3CCE, 2020-07-14:
From the develment branch released by Zarach 9 hours ago.

zarath, 2020-07-14:
@PA3CCE 
it's not the latest

I see it in the hardware detection.

Do you have an -H -H4 variant, too?

Latest is https://github.com/NanoVNA-Saver/nanovna-saver/releases/tag/v0.3.6

PA3CCE, 2020-07-14:
@ igor-m.... I only have the -H4..... Your link directs me to the one I downloaded !!!

zarath, 2020-07-14:
Exactly this? https://github.com/NanoVNA-Saver/nanovna-saver/releases/download/v0.3.6/NanoVNASaver.windows-latest.zip

igor-m, 2020-07-14:
> 
> 
> @ igor-m.... I only have the -H4..... Your link directs me to the one I downloaded !!!

@PA3CCE: I have not provided any link to you.. I've been just asking you kindly to tell us where did you download the repository from, and I do it for some time already :)

zarath, 2020-07-14:
the freeze occurs as H4 didn't respond with possible bandwidth settings to a plain bandwidth command.
Original NanoVNA tells about possible settings if that command is called without parameters.
In current version I cannot do bandwidth settings for -H versions ... :-(
That's the current workaround.
I -H4 is able to do a bandwidth equivalent to 10 as in the original NanoVNA and is set to this at connection, i expect  trouble...
:-(
And as long as I didn't find the current source of DiSlords patches I cannot reverse engineer the behaviour of -H versions.

PA3CCE, 2020-07-14:
@ igor-m.....Oh sorry it was from zarath....., I expected response from you.... Zarath commented the it was not the latest and gave me a link that directed me to the one I downloaded.

@Zarath the link you gave me first was directing me to the one I downloaded.....The second one does not work, it tries to connect but it aborts.

zarath, 2020-07-14:
PA3CCE can you try to install from source......
it's just installing python from the microsoft store and then follow the installation instructions in the readme

PA3CCE, 2020-07-14:
@Zarath.....The first link you gave me was ok it directed me exactly to where I downloaded the NanoVNASaver.windows-latest.zip file.
I have Python 3.7 and 3.8.3 installed on my D: drive but really dont know how exactly to compile your source files, vy sorry.

zarath, 2020-07-14:
Nothing to compile. Python is an interpretated language

PA3CCE, 2020-07-14:
OK, I understand, in the very past I have worked with interpreters and Pascal programming, but I installed Python but was too busy with other things to study how to work with it.... It's getting rather late now, so I would continue tomorrow if you don't mind.

PA3CCE, 2020-07-14:
@ Zarath.... I just read your comments abt the bandwith settings, For your info, I only use the -H4 version and the bandwith setting was 100 Hz.... See you tomorrow if you are here.... Good night to you and igor-m.

PA3CCE, 2020-07-15:
@Zarath  I foud your message with the second link to the last version of NanoVNA-Saver in my e-mail as well. From there I was able to download the Windows.zip file. I'm sure that I had the correct one, but I will install the program again from this last download and the run it with the bandwidth set to 1kHz,  will let you know the result.

PA3CCE, 2020-07-15:
@Zarath...NanoVNA -H4 to  ON.... Device set to Bandwidth 1kHz....Deleart/stopted  repository NanoVNASaver from C: drive:.... Deleted NanoVNA-Saver 3.6 rc from my D: drive. ....Re-installed NanoVNA-Saver 3.6 rc from the **last** download...... Opened a  Command Window.....In command window went to repository  NanoVNA-Saver 3.6 rc on my D: drive..... executed nanovna-saver.exe -D debug.txt....Program showed up with empty start/stop and 1 segments, Com port: Com11..... pressed "Connect to device"...........One sweep was performed......Segments set to 10......Start/stop set to 50MHz/2GHz....., Segments set to 10................Sweeps were performed correctly with resonable results......Loaded a corresponding calibration file made by version 3.4.......PROGRAM ABORT ! .......Searched for debug.txt file, not present in NanoVNA-Saver 3.6 rc repository..........Command window (not aborted) shows at the end :

Settings: C:/users/jrs/AppData/Roaming/NanoVNASaver/NanoVNA.ini
Traceback most recent (call last):
File "NanoVNASaver\Windows\CalibrationSettings.py", line 118, in <lambda>
File "NanoVNASaver\Windows\CalibrationSettings.py", line 575, in loadCalibration
File "NanoVNASaver\Windows\CalibrationSettings.py", line 455, in calculate
AttributeError: 'NanoVNASaver' oblect has no attribute 'btnStopSweep'

Will do some more testing

PA3CCE, 2020-07-15:
@Zarath Sorry in the first line "Deleart/stopted" must be "deleted"

PA3CCE, 2020-07-15:
@Zarath...After the above,  in the command window "nanovna-saver.exe -D debug.exe" does start the program but clicking "Connect to device"  causes freezing of the program, like it did before using the old installation, even after deleting the .ini file on the C; drive. 

What has changed where ??

zarath, 2020-07-15:
freeze is caused by bandwidth detection. It's different between ttrfech's and dislord's firmware.

I've flashed my nanovna with current dislords one and hopefully fixed it in -rc1

PA3CCE, 2020-07-15:
OK, it's clear to me, in the rc phase you have to de- re-install the program after aan error.

zarath, 2020-07-15:
that shouldn't be necessary

[31 more replies]

closed 2020-07-15 by zarath