NanoVNA-Saver/nanovna-saver #245: Touchstone Export File Format Bug

Roger-Need, 2020-07-14:
I am using NanoVNA Saver 0.3.4 and found a bug with Touchstone file export.   There is a formatting bug that prevents the file from being loaded into other programs.  It has to do with the Option Line.  The frequency is expressed in HZ and that is not valid.  It must be Hz  (z is lower case).  This seems to have changed since 0.2.2 which worked OK.  

I attached an excerpt from the Touchstone specification below...
[image]

zarath, 2020-07-14:
What about the first line in the 1.1 Touchstone definition?

[image]

Touchstone is defined as case insensitive.

What program is complaining about the option line?

Roger-Need, 2020-07-14:
The program is AntScope2.   [https://rigexpert.com/products/software/antscope2/](url)

This program developed by RigExpert now supports the NanoVNA-H and there is a planned release for other NanoVNA models.   It is beneficial to be able to exchange .s1p files with this program because it supports multiple S11 traces, de-embedding cable feedlines, conversion to other system impedances and several other useful features.  

I can manually go in and edit HZ to Hz and it works but other users will probably not know this and just assume a file incompatibility.

Roger-Need, 2020-07-14:
Oops - closed by mistake....

zarath, 2020-07-14:
see https://github.com/rigexpert/AntScope2/issues/6

Roger-Need, 2020-07-14:
Zarath - I see you posted this issue on the AntScope2 github.  That will not help solve the problem.  RigExpert basically just ignore issues on github and are not very responsive to feedback on their own groups.io forum.  

 Is it a big change to put it back the way it used to be with "Hz" in 0.2.2?

zarath, 2020-07-14:
It could break imports in other broken software....

Fun fact: RigExpert use scanf and strcmp with qt and c++ in 2020!!!!

Roger-Need, 2020-07-15:
I have edited touchstone.py to output Hz and built an executable version of NanoSaver 0.3.4.  It now exports files that are importable by other popular programs (AntScope2, AppCad, Filpal, VNWA, NanoVNA Mod V3, NanoVNA#, NanoVNA Partner).  Since this issue is now marked "invalid" I will fork NanoVNA Saver to another github and release versions from there for those interested.   A link to a Windows executable should be available on groups.io and Facebook in a few days...

Roger

closed 2022-09-18 by zarath