NanoVNA-Saver/nanovna-saver #251: Wrong Center and Span Frequency 036-rc

igor-m, 2020-07-14:
Wrong Center and Span Frequency in 036-rc, see below (H4 0845fw).
[image]

zarath, 2020-07-14:
Wrong or just not fitting?

zarath, 2020-07-14:
I assume there is a 1.899Ghz span

igor-m, 2020-07-14:
It could be the Span does not fit, but the Center is definitely not good..

zarath, 2020-07-14:
Have you clicked in and used cursor keys to scroll left and right

zarath, 2020-07-14:
There are anti alias artifacts that suggest that the 0 is not the first digit

igor-m, 2020-07-14:
You are right it does not fit..
[image]

zarath, 2020-07-14:
950,025MHz would be center... thus 0025 looks plausible

igor-m, 2020-07-14:
That is because I do not own a 4k display..

zarath, 2020-07-14:
It was annoying me long time ago, but me PR was rejected.
I'll look for a better Input field arrangements in future. But definitely nothing to go in 0.3.6... Gui quirks are so depending on individual settings. And as long as markers and scales are right I didn't bother too much.

Also it's not a 0.3.6 introduced bug

igor-m, 2020-07-14:
Simply round to 3 digits after decimal point, the high precision of that information is not critical imho.

zarath, 2020-07-15:
This is a duplicate of #154

closed 2022-11-10 by github-actions[bot]