NanoVNA-Saver/nanovna-saver #252: Divide by zero after Connected to device

igor-m, 2020-07-14:
Many attempts to start the v036rc end up with a crash, where the H4 0845fw crashes as well (needs off/on).

From CMD:

```
..
Settings: C:/Users/me/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
2020-07-15 00:10:16,585 - NanoVNASaver.SweepWorker - ERROR - division by zero
Traceback (most recent call last):
  File "NanoVNASaver\SweepWorker.py", line 125, in run
  File "NanoVNASaver\SweepWorker.py", line 172, in _run
  File "NanoVNASaver\SweepWorker.py", line 297, in readAveragedSegment
  File "<__array_function__ internals>", line 5, in average
  File "site-packages\numpy\lib\function_base.py", line 381, in average
ZeroDivisionError: division by zero
Traceback (most recent call last):
  File "NanoVNASaver\SweepWorker.py", line 128, in run
  File "NanoVNASaver\SweepWorker.py", line 125, in run
  File "NanoVNASaver\SweepWorker.py", line 172, in _run
  File "NanoVNASaver\SweepWorker.py", line 297, in readAveragedSegment
  File "<__array_function__ internals>", line 5, in average
  File "site-packages\numpy\lib\function_base.py", line 381, in average
ZeroDivisionError: division by zero
```

and attached you may find the debug log.
[debug036Q.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4921777/debug036Q.txt)

zarath, 2020-07-15:
Error in data acquisition should not crash the program

igor-m, 2020-07-15:
Also - where this floating point comes from?
..
2020-07-15 00:10:14,875 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 41.801980198019805
2020-07-15 00:10:14,938 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
..

zarath, 2020-07-15:
If forgotten to round them... doesn't matter ...

A retry takes 0.05 seconds.
I expect to get all data after a maximum of retries :-)

That value is calculated by following simple formula....

def _max_retries(bandwidth: int, datapoints: int) -> int:
    return 20 * (datapoints / 101) + round(
        (1000 / bandwidth) ** 1.2 *  (datapoints  / 101))

exponent was estimated by doing measurements with different bandwidth
lower bandwidth - > longer time

zarath, 2020-07-31:
can you try rc03

closed 2020-07-31 by zarath