NanoVNA-Saver/nanovna-saver #257: Frequency sweep setting for 40M is incorrect

kzembower, 2020-07-15:
When choosing 'Sweep settings..." and then "Sweep band" "Select band..." "40M", the limits are 7.0 to 7.2MHz. In the U.S. the band goes from 7.0 to 7.3MHz.

OS: Linux Ubuntu 18.02 LTS
Python 3.7
NanoVNA-Saver version:  0.3.4

This is not a big deal. Just reporting to help you find and squash all the bugs.

-Kevin
KC3KZ

zarath, 2020-07-15:
Bands are configurable in "display setup"-> "manage bands"

Author of nanovna-saver is living in ITU region 1 :-)

kzembower, 2020-07-15:
Thanks. I didn't know either of those two facts. I'll just adjust my 40M 
band limits.

Thanks for pointing those things out. You can consider this bug report 
'closed.'

-Kevin
KC3KZ

closed 2020-07-16 by zarath