NanoVNA-Saver/nanovna-saver #258: Data outside the Smith Chart

igor-m, 2020-07-15:
Both nanovna meter as well as the nanovna-saver can depict the data outside the Smith Chart.
I asked on that issue (group.io) and got that it depends on the calibration (quality of the cal set, or changes made after calibration, ie. different cable, etc).
I think technically it could be pretty easy to create a mess with the calibration and the measurement, the Smith Chart should not show data outside its border, as the space outside the SC border "does not exist".
I would suggest either to
a) not show such outliers, or,
b) place the outliers at the SC border.
Below an example.
[image]

zarath, 2020-07-15:
It's just data behind infinity... .it falls of the border of the earth disk.

But I'm uncertain if this values, which are created by cable, plug, calibration inaccuracies should be suppressed.

igor-m, 2020-07-15:
My math knowledge gets pretty rusty, but I am pretty confident that "with _any_ values of the calibration data the measurement results will _always_ be inside the SC space".

igor-m, 2020-07-16:
FYI - "3D Smith Chart" tool -  https://www.microwavejournal.com/articles/33830-d-smith-chart-tool-is-now-free-to-all-students

satfan52, 2020-07-18:
Normally of the calibration is done with a good calibration kit,this should not be not an issue in practice. 

The other case that I have encountered is a reflexion coeficient greater than 1 and a negative real value of an input impedance.  This was when I measured an unstable amplifier that tended to oscilliate at some frequencies, but that measurement was entirely normal and to be expected with unstable active device

igor-m, 2020-07-24:
The Smith chart cannot show data outside its border. It is like an o'scope screen..  :)
Imagine yourself an oscilloscope screen - you may do whatever you want with its settings, probes calibration, etc., it will not show the traces outside its screen.

satfan52, 2020-08-27:
Actually it can with an unstable amplifier but not with passive devices (filters etc..). Check the literature on stability factors and stability circles which all have to do with reflection coeficient bigger than 1 and falling outside the smithchart. Myself, I have experienced that some amplifiers that I build/test can  exhibit a reflection coefficient bigger than 1 at low frequencies, the nanovna is not wrong.

closed 2022-11-10 by github-actions[bot]