NanoVNA-Saver/nanovna-saver #259: Calibration - Offset Delay - Setting

RandMental, 2020-07-16:
Saver uses a decial point for values, but the Offset Delay in the Calibration dialogue  use a decimnal comma.

[image]

zarath, 2020-07-16:
looks like languages settings in qt...

Nothing that I've ever touched... First occourance in 0.3.6????

zarath, 2020-07-16:
Depends on LC_NUMERIC environment in linux ... with German I get a comma unset I get a point

zarath, 2020-07-16:
Funny, in South Africa both decimal separators are possible...

https://en.wikipedia.org/wiki/Decimal_separator#Countries_using_decimal_comma

So I think a recommendation to use a US or GB locale with nanovna-saver to get a consistent view ...

closed 2022-11-10 by github-actions[bot]