NanoVNA-Saver/nanovna-saver #260: NanoVNA-Saver 3.6 rc2 shows markerdata when NOT set for

PA3CCE, 2020-07-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start NanoVNA-Saver.
2. Click on "Connect to device".
3. NanoVNA-Saver shows data from stopfrequency in Markerdata-area (Incidently I have seen data from startfrequency).
4. See error in attached picture
5. Debug file is attached

[image]

[dbug-rc2-2.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4933703/dbug-rc2-2.txt)

**Expected behavior**
No data in markerdata-area

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows10 Home edition.
 - Python version: NA
 - NanoVNA-Saver version: 3.6 rc2

igor-m, 2020-07-16:
That is not a bug, rather a feature - like default settings with empty Marker's fields.
With empty fields it would be nice to have the markers filled with frequencies like, for example, in case of 5 markers
f=(stop-start)/4
M1 = start
M2 = M1 + f
M3 = M2 + f
M4 = M3 + f
M5 = stop

PA3CCE, 2020-07-17:
OK, I agree, but at the moment they are filled with all the same data. 5 markers is maximum, otherwise window is oversized.

PA3CCE, 2020-07-17:
PLease why Zarach ?

zarath, 2020-07-17:
because illegal values in enabled markers causes more trouble and lead to more complex and therefore more error prone code

closed 2020-07-17 by zarath

PA3CCE, 2020-07-17:
OK, then leave the areas open, please.

PA3CCE, 2020-07-17:
Or disable the markers.