NanoVNA-Saver/nanovna-saver #265: Program stops after a few minutes

PA0JOZ, 2020-07-17:
good evening,
The programs stops after a few minutes of continuous sweeping.
Using version 3.6 under Windows 10 with the nanoVNA V2 (S-A-A V2).
After a few minutes of continuous sweeping  the sweeping stops and a few seconds later the program windows disappears from the screen. Restarting can only be done after powering down the S-A-A V2 and powering it up again.
Regards, 73, Jos, PA0JOZ

zarath, 2020-07-17:
Please try with debugging. Nanovna.exe -D debug.txt and attach debug file after crash here.

PA0JOZ, 2020-07-18:
Good afternoon Holger,
Very strange, yesterday evening the application crashed all the time after
a few minutes. Today I do not seem to be able to reproduce that behaviour.
I tried many times and it crashed only once. Unfortunately the debug
function was not on at that instance. But I made a screenshot of the
command window after the crash. See the attached screenshot. I hope it
helps.
Best Regards, 73, Jos, PA0JOZ

zarath, 2020-07-18:
Hi Jos,
sometimes errors in the usb communication occur.
This leads to exceptions in the sweep worker of nanovna-saver.
I'll try to catch them in future releases, so that at least the program wouldn't crash.

closed 2020-07-30 by zarath