NanoVNA-Saver/nanovna-saver #266: Implement basic de-embeding functionality in nanovna saver

satfan52, 2020-07-18:
Sometimes it is not possible or even desirable to measure a DUT directly. Instead what can be measured is

measurement 1:  network 1-DUT-network2
measurement 2:  network 1
measurement 3:  network 2  

Based on these 3 measurements, it is then possible to calculate the DUT S parameters with a bit of math.

This is done by converting the 3 S matrixes  (S1,S2, S3) corresponding to the 3 measurements above into T ones  (T1,T2,T3). Then the T matrix of T1 and of T3 are inverted and the T matrix of the DUT is calculated as follow : T_DUT= INV(T2)*T1*INV(T3) Finally, the S matrix of the DUT is calculated by inverting the T matrix of the DUT. There is nothing new, all of this is very well known and described at length on the Internet, notably by Keysight and ADI
.
https://www.keysight.com/be/en/assets/7018-06806/application-notes/5980-2784.pdf
http://na.support.keysight.com/vna/help/latest/S3_Cals/Fixturing.htm#deembed

It is proposed to offer the possibility to import in Nanovna, in the File Menu, as touchtone SP2 files, the S2 and S3 files corresponding to the network 1 and network 2. When nanovna saver detects that those files have been imported, it would then do the calculation cited above at every sweep in order to do the de-embeding. 

The change needed to the GUI FILE interface would be the following :
"import S2P files as network 1"
"import S2P files as network 2"
"check-box to enable/disable de-embeding at every sweep"

It is assumed that the imported S2P files are populated with S11 and S21 only of course. The same operation can be created by reversing the DUT to get deembeded S22 and S12. Then if #203 is implemented too, it is then possible to export a fully populated S2P files with the 4 de-embeded parameters of the DUT. This file can then be analyzed in third party software like SimSmith for instance.

igor-m, 2020-07-19:
I've been coping with a bare wire antenna measurement.. Such a measurement would require de-embedding, imho.
Btw, a "plug-in" nano-saver functionality would be "nice to have", thus people would be able to write special procedures and math.

satfan52, 2021-03-13:
In the mean time, the SIMSMITH software has been upgrade to provide this functionality, **so if the 4 S parameters can be exported in an S2P file it is actually OK**, it makes this change request much less urgent.   Some functionality is required to captre S12, S21 but also S22 and S12 (after reversing the device) and the producing a fully populated S2P file with the 4 parameters

human890209, 2022-07-15:
> 

How to do this with SimSmith( current version: SimNEC 1.2)?

human890209, 2022-07-15:
> In the mean time, the SIMSMITH software has been upgrade to provide this functionality, **so if the 4 S parameters can be exported in an S2P file it is actually OK**, it makes this change request much less urgent. Some functionality is required to captre S12, S21 but also S22 and S12 (after reversing the device under test) and the producing a fully populated S2P file with the 4 parameters

How to do this with SimSmith( current version: SimNEC 1.2)?

Ho-Ro, 2023-01-17:
Please check if also #203 has relevance.

closed 2023-06-22 by github-actions[bot]