NanoVNA-Saver/nanovna-saver #267: Marker 2 colorsetting not possible

PA3CCE, 2020-07-18:
**Describe the bug**
Marker 2 color can't be set..

**To Reproduce**
Steps to reproduce the behavior:
1. Click on color square of marker 2 in main window.
2. Set for color to be used in the color selection window.
3. Click "ÖK".
4. New color is not set.

**Expected behavior**
New color for marker 2 is set.

**System** :
OS : Windows 10 Home edition.
Device : NanoVNA -H4 firmware 0.9.3.4 by DiSlord.
NanoVNA-SWaver version 3.6

zarath, 2020-07-21:
Unreproducable on my machine

PA3CCE, 2020-07-21:
Tried this again also, the color square still is not filled with the selected collor.

PA3CCE, 2020-07-21:
OK, Started NanoVNA-Saver 3.6 connected it to the -H4 and first tried to set marker 2  color, no success then  tried to set it on the trace , no result either, maybe because there is not set a color ?..... Hope you find something in the debug_Marker_2 file
[debug_Marker_2.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4953063/debug_Marker_2.txt)

PA3CCE, 2020-07-21:
@Zarach Problem sol;ved, please see my last comment in **Marker 2 does not show in display**

closed 2020-07-21 by PA3CCE