NanoVNA-Saver/nanovna-saver #268: Marker 2 does not show in display

PA3CCE, 2020-07-18:
**Describe the bug**
Marker 2 does not show in display when set for.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on circle marker 2 in main window.
2. Click on place where to set the marker on the trace.
3. Marker does not show up.

**Expected behavior**
Marker 2 shows on trace in the display.

**Desktop (please complete the following information):**
 - OS: [e.g. : Windows10 Home edition.
 - Python version: NA
 - NanoVNA-Saver version: V 3.6
Add any other context about the problem here.

PA3CCE, 2020-07-21:
Just checked it several times, still have the problen here.
Will do it from cmd window, but wonder if the debug file shows anything.

PA3CCE, 2020-07-21:
When I klick marker 2 on the trace it does'nt show there, freauency bar is being filled. filled

RandMental, 2020-07-21:
Have erased your .ini file?  Make sure you start all afresh.

PA3CCE, 2020-07-21:
@RandMental .... No I did NOT , To be sure I will do so !
@ Zarath.... At start of NanoVNA-Saver NO default color is set !..... I started NanoVNA-Saver from command window , first tried to set the color , no result , then tried to set the marker on the trace , no result. Hope the debug_Marker_2 file will he;p !
[debug_Marker_2.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4953138/debug_Marker_2.txt)

PA3CCE, 2020-07-21:
**Deleted the NanoVNASaver repository on my C: drive**.....Started NanoVNA-Saver 3.6 from  command window had to answer wellknow questions and had to set my preferred settings again....., clicked on color square from marker 2 (was filled)..... set to for new color, .....pressed OK..... new color was filled in square.....  clicked on trace, in the display....  marker 2 shows on trace in newly  set color...........**Both problems are solved.**

@ RandMental **Thank you so much.**

@ Zarach.....**Adding a warning to delete the repository after updating to a new version of NanoVNA-Saver looks recommendable.**

RandMental, 2020-07-21:
Great, happy your markers are now back,

closed 2020-07-21 by zarath