NanoVNA-Saver/nanovna-saver #270: Release 0.3.7 pre - Nanovna V2 support - Calibration - Multiple segments - Divide by zero and crash

satfan52, 2020-07-20:
Tested with the nanovna-v2  and nanovna-saver 0.3.7-pre

In release 3.4 it was possible to do a calibration with 10 segments of 101 points, it was working quite well. During the measurement sweep, I was selecting either 1 or 10 segments depending on the speed/resolution tradeoff  coresponding to my needs.

In the current development version, attempting to do a calibration with 10 segments crashes the application
 
root@on7yi-ubuntu:/opt# nanovna-d
NanoVNASaver 0.3.7-pre

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Settings: /home/peter/.config/NanoVNASaver/NanoVNASaver.ini
2020-07-20 13:33:56,838 - NanoVNASaver.Hardware.NanoVNA_V2 - ERROR - expected 3232 bytes, got 3136
2020-07-20 13:33:56,840 - NanoVNASaver.SweepWorker - ERROR - division by zero
Traceback (most recent call last):
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 125, in run
    self._run()
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 173, in _run
    start, stop, averages)
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 297, in readAveragedSegment
    values11 = np.average(values11, 0).tolist()
  File "<__array_function__ internals>", line 6, in average
  File "/usr/local/lib/python3.7/dist-packages/numpy/lib/function_base.py", line 391, in average
    scl = avg.dtype.type(a.size/avg.size)
ZeroDivisionError: division by zero
Traceback (most recent call last):
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 128, in run
    raise exc
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 125, in run
    self._run()
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 173, in _run
    start, stop, averages)
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 297, in readAveragedSegment
    values11 = np.average(values11, 0).tolist()
  File "<__array_function__ internals>", line 6, in average
  File "/usr/local/lib/python3.7/dist-packages/numpy/lib/function_base.py", line 391, in average
    scl = avg.dtype.type(a.size/avg.size)
ZeroDivisionError: division by zero

zarath, 2020-07-25:
can you try current Development tree?

satfan52, 2020-07-31:
Hello,
I did the test with the development branch and nanovna-saver does not crash. The sweep of 2020 points completes fine unlike with the master branch (crash happens  at about 60% with the master branch). 

However, there are very strange errors with the development branch. See console output below

[image]

[image]

[image]

root@on7yi-ubuntu:~# nanovna-d
NanoVNASaver 0.3.7-rc03

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Settings: /home/peter/.config/NanoVNASaver/NanoVNASaver.ini

2020-07-31 12:48:01,265 - NanoVNASaver.Hardware.NanoVNA_V2 - ERROR - expected 6432 bytes, got 5536

2020-07-31 12:48:13,382 - NanoVNASaver.Hardware.NanoVNA_V2 - ERROR - expected 6432 bytes, got 5952

zarath, 2020-07-31:
Perhaps you have done a calibration with 2 segments á 101 points.
It will work as long as you inside the calibration boundaries, as cal data will be interpolated if a frequency is measured between cal points.

That ERROR expected should not crash the program as long as it does not happen to often during on sweep segment.
If this error occurs the sweep repeats the scanning of that segment... perhaps i can level the log down to warning, as
this happens more often than I expected.

zarath, 2020-07-31:
Regarding crashing, can this bug be closed?

satfan52, 2020-07-31:
yes please

satfan52, 2020-08-11:
Unfortunately, I was too quick to close this ticket. 

a) Yes  I can do a 10 segment sweep of 101 points per segment with the nanovna V2 and with the latest nanovna-saver development version.  But this will only work if I have calibrated with 101 points. 

b) What does NOT work is a 10 segment CALIBRATION  of 101 points per segment,  so 1010 points total. The crash during the calibration procedure  is systematic and very easy to re-produce.  I am using the Wizard/Assistant to do the calibration.

c) Intersingly, please note that a one segment calibration of 1021 points (select 1021 points under manage device) WORKS of course. However what does NOT work, is to do a 1 segment sweep of 101 point after this calibration of 1021 points (ie, change from 1021 to 101 points under managed device), even after stoping and restarting the application. I get the very SAME error
  
d) All of this works very well with the nanovna V1, so this problem is V2 specific

Objective: The idea is to do the calibration with a one segment sweeps of 1021 points  or with 10 segment sweeps of 101 points/segment. Then depending on   how long I am willing to wait during my measurements and depending on the resolution I want, I do a 1 segment sweep  of 101 points, multiple segment sweeps of 101 points/segment  or eventually a 1 segment sweep of 1021 points. 

[image]

root@on7yi-ubuntu:/opt/nanovna-saver-d# nanovna-d
NanoVNASaver 0.3.7-rc07

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Settings: /home/peter/.config/NanoVNASaver/NanoVNASaver.ini
2020-08-11 12:30:35,193 - NanoVNASaver.Hardware.NanoVNA_V2 - ERROR - expected 3264 bytes, got 3168
2020-08-11 12:30:53,792 - NanoVNASaver.SweepWorker - ERROR - argument out of range
Traceback (most recent call last):
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 80, in run
    self._run()
  File "/opt/nanovna-saver-d/NanoVNASaver/SweepWorker.py", line 139, in _run
    self.app.vna.resetSweep(start, end)
  File "/opt/nanovna-saver-d/NanoVNASaver/Hardware/NanoVNA_V2.py", line 183, in resetSweep
    self.setSweep(start, stop)
  File "/opt/nanovna-saver-d/NanoVNASaver/Hardware/NanoVNA_V2.py", line 222, in setSweep
    self._updateSweep()
  File "/opt/nanovna-saver-d/NanoVNASaver/Hardware/NanoVNA_V2.py", line 228, in _updateSweep
    int(self.sweepStartHz - (self.sweepStepHz * s21hack)))
struct.error: argument out of range

zarath, 2020-08-11:
It's a different error, please supply debug file.

with that small dump i cannot see why it wasn't catched.

That expected ... bytes error seems to be a timing issue on vna's hardware. it's happening regullary and I normally do a retry when this happens.

closed 2020-08-13 by zarath