NanoVNA-Saver/nanovna-saver #271: pyQT Issues on Ubuntu 20

sprakashb, 2020-07-20:
I was trying to run the v 0.3.6 (64 bit from release ) on ubuntu 20 and it gave following report

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

I tried installing a fresh copy of pyQT5, but same error persisted.

zarath, 2020-07-20:
Strange -- I use ubuntu 20 as development environment. But if I exit my venv and unzp the binary, chage to the directory, chmod +x ./nanovna-saver and call it as ./nanovna-saver it runs flawlessly.

zarath, 2020-07-21:
Unreproducable on my machine

igor-m, 2020-07-22:
I got that issue on the Ubuntu 18.04 during install.
I had to reinstall the libxcb package and it worked afterwards..

sprakashb, 2020-07-22:
Friends,
I tried a fresh installation of Ubuntu 20 and it worked. Fantastic piece of
software.

In the other system the error is still there.

Will try the suggestion.

Thank you all for your help.

Vu2spf
SP Bhatnagar

closed 2020-07-24 by zarath

sprakashb, 2020-07-24:
Hi OM Holger,

I will be talking on Nano VNA on a HAMS group online where I have
specifically mentioned and showing the NANO-VNA_SAVER. It would be on
coming Sunday (26th July 2020) at 830AM Indian Standard Time (5Hr30m Ahead
of UTC). If you or anyone would like to join please let me know and I will
send the link once I get it. It would be on YouTube Live.

Best regards,

VU2SPF
SP Bhatnagar

If you would

zarath, 2020-07-24:
> Hi OM Holger, I will be talking on Nano VNA on a HAMS group online where I have specifically mentioned and showing the NANO-VNA_SAVER. It would be on coming Sunday (26th July 2020) at 830AM Indian [...]

Nice to hear, but that's 5 o'clock local here ... I think, I need my sleep then...

sprakashb, 2020-07-24:
Ok with that. I think they have a link for viewing later. If that is
available I shall send it to you.

Best regards,

Vu2spf