NanoVNA-Saver/nanovna-saver #272: MJissing things in some displays (Saver 0.3.6)

PA3CCE, 2020-07-20:
**Describe the bug**
In some S11 and S21 displays some items are missing..

**To Reproduce**
Steps to reproduce the behavior:
1. Load and start NanoVNA-Saver 3.6
2. Click on  "Display setup" and add one or more markers.
3. Run a  sweep.
4. Press "Display setup ..."
5. Select a  "Displayed charts / **Real & Imaginary" "**
6. The SWR marker line(s) is/are not shown, shall be there between 0 and -10 db lines.
7. Press "Display setup...".
8.  Select a "Displayed charts/ **S11 Real/Ímaginary".**
9. In the respective chart **no** color indication is given **what** trace is **what** color.
10. Press "Display Setup".
11. Select a "Displayed charts/ **S21 Real/Imaginary"** .
12 .In the respective chart **no** color indication is given **what** trace is **what** color.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows10 Home edition.
 - Python version:  NA
 - NanoVNA-Saver version:  0.3.6
[image]

zarath, 2020-07-21:
Also not been in v0.2.x versions. an older bug

PA3CCE, 2020-07-21:
About version 0.2.x, yes I know, but imho these things must be there.

zarath, 2020-07-22:
I've just remove the reference to 3.6 as it is existing on all versions. So I don't have to do researching if I introduced the bug in my recent code changes.

PA3CCE, 2020-07-25:
ok,  Iunderstand

closed 2022-11-10 by github-actions[bot]