NanoVNA-Saver/nanovna-saver #277: NanoVNA crashes or wrong data displayed after first start of Nanovna V2

ok1vaw, 2020-07-23:
After connecting NanoVNA V2 (without battery inside) to USB and starting NanoVNA Saver the program crashes or shows incorrect values. It needs to reconnected and connected again, after that it works. Tested with different USB2 and USB3 port and different cables, appears on version 034, 035, 036, USB2 direct to PC or from HUB with power supply. Debugging info attached and screenshot with strange measurement added.
[debug_instability.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4969561/debug_instability.txt)
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4969562/debug.txt)
[debug_instability.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4969568/debug_instability.txt)
[image]

zarath, 2020-07-31:
can you try v0.3.7-rc03 or newer

ok1vaw, 2020-08-15:
Still the same, I have even tried a second USB PCI card with NEC chip (AXAGO PCIU-60), but it behaves in the same way.  Now I test it with the RC05.

ok1vaw, 2020-08-21:
With installing the Li-Ion battery inside, this problem with start disappeared.

Mike4U, 2020-10-19:
See #322  at Mike4U

closed 2022-01-06 by zarath