NanoVNA-Saver/nanovna-saver #281: NanoVNA-Saver 3.7 pre01 does not see the device

PA3CCE, 2020-07-25:
**Describe the bug**
NanoVNA-Saver 3.7 pre01 does not see my -H4

**To Reproduce**
Steps to reproduce the error
1. Downnloaded nanovna-saver-Development.zip from development branche and extracted it to repository X
2. Opened command window and navigated to repository X
3. Executed command  "pip3 install ." 
4. Got "Successfuly instaled  NanoVNASaver 3.7rc1" reported
5. Executed command  "nanovnasaver".
6. NanoVNA-Saver 3.67rc01 is loaded, but "connect to device" text bar is **not filled** (while NanoVNA-H4 is connected to a USB           port)
7. Pressed  "Rescan",  "connect to device"  text bar is **not filled**
8. Cannot connect to device.

**Expected behavior**
Can connect to my NanoVNA-H4
.
**Desktop (please complete the following information):**
 - OS: Windoes 10 Home edition
 - Python version: Python 3.7 and 3.8
 - NanoVNA-Saver version: 3.7rc01

**Additional context**
Add any other context about the problem here.

zarath, 2020-07-25:
can you try the binary from the asset link?

PA3CCE, 2020-07-25:
OK, the asset link shows **no** binary just the **source** files,  I downloaded the sourcode.zip and after extracting and installing this one it "sees"  the device on com11 and I can run it, however on the top it still shows NanoVNA-Saver 0.3.6. is this correct ?

PA3CCE, 2020-07-25:
The previous installation showed 3.7pr01 in the top, anyhow I installed the one from the asset link and executed a swerep from 3MHz to 2GHz on a BiasTee , using  a calibration from version 3.6 and 10 sgegments,  bandwidth set to 10 Hz, reasonable result, see attachment. If it's really version 3.7 how to use the added marker items, I can' find them.
[image]

RandMental, 2020-07-26:
Nope, you are running the "older" version

PA3CCE, 2020-07-26:
@ RandMental.... Yes, I was afraid so, but Zarath advised me to download from the asset link, where do I find the latest one ?

igor-m, 2020-07-26:
@PA3CEE: when you run on WIN10 - install zarath's git repo on your local disk and simply do anytime

`Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Users\smith> cd C:\Users\smith\MyCode\HAMradio\NanoVNASaver\nanovna-saver_zarath
PS C:\Users\smith\MyCode\HAMradio\NanoVNASaver\nanovna-saver_zarath> cd nanovna-saver
PS C:\Users\smith\MyCode\HAMradio\NanoVNASaver\nanovna-saver_zarath\nanovna-saver> git fetch
..
PS C:\Users\smith\MyCode\HAMradio\NanoVNASaver\nanovna-saver_zarath\nanovna-saver> git pull
..
PS C:\Users\smith\MyCode\HAMradio\NanoVNASaver\nanovna-saver_zarath\nanovna-saver> ./nanovna-saver.py -D debug037pre01.txt
PS C:\Users\smith\MyCode\HAMradio\NanoVNASaver\nanovna-saver_zarath\nanovna-saver>`

and you get the latest nanovan-saver running..

PA3CCE, 2020-07-26:
@ Zarach.... This afternoon I downloaded the source files from the development branch again, instaled an excecuted the program, this time the device was recognised and I could connect it. I ran a sweep from 3MHs to 2GHz with Segments set to 10 and Bandwith set to 10 Hz, I used a calibration file made by version 3.6. The sweep ended with an error message : " Error during sweep, Stopped, list assignment index out of range" .... However the sweep ended correctly (see attacment).

[image]

@igor-m Thanks, but I succeded as described.

PA3CCE, 2020-07-26:
@ igor-m.... This is what I did : Downloaded the source files from the development branche, loaded them in  repository x, loaded a command window , navigated to repositort x, executed command "pip3 install ." , then " nanovnasaver" this loads the program.

zarath, 2020-07-26:
I'll close this issue, as the main problem with the list index is already mentioned in an other open issue, and the topic of this bug is gone

closed 2020-07-26 by zarath