NanoVNA-Saver/nanovna-saver #283: Failed reading data on NanoVNA-H

RexAlfes, 2020-07-26:
I've tried V0.3.6 and 0.3.7. Error reading data. The following screen dump is from pressing the connect button. The scan works and sees the NanoVNA on my USB2 and USB3 ports. I have a brand new NanoVNA-H.

NanoVNASaver 0.3.7-pre02

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Settings: C:/Users/ralfe/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
2020-07-26 07:18:20,842 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:21,241 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:21,639 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:22,039 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:22,437 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:22,639 - NanoVNASaver.SweepWorker - ERROR - Tried and failed to read data 0 5 times.
2020-07-26 07:18:22,939 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:23,341 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:23,739 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:24,137 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:24,534 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (2147483647.147483647 2147483647.147483647)
2020-07-26 07:18:24,735 - NanoVNASaver.SweepWorker - CRITICAL - Tried and failed to read data 0 10 times. Giving up.
2020-07-26 07:18:24,735 - NanoVNASaver.SweepWorker - ERROR - Failed reading data 0 10 times.
Data outside expected valid ranges, or in an unexpected format.

You can disable data validation on thedevice settings screen.
Traceback (most recent call last):
  File "NanoVNASaver\SweepWorker.py", line 84, in run
  File "NanoVNASaver\SweepWorker.py", line 133, in _run
  File "NanoVNASaver\SweepWorker.py", line 245, in readAveragedSegment
  File "NanoVNASaver\SweepWorker.py", line 268, in readSegment
  File "NanoVNASaver\SweepWorker.py", line 314, in readData
OSError: Failed reading data 0 10 times.
Data outside expected valid ranges, or in an unexpected format.

You can disable data validation on thedevice settings screen.

RexAlfes, 2020-07-26:
Problem resolved after a full calibration on the NanoVNA.

zarath, 2020-07-26:
Ok. Personally I do not see much sense in the Validate Data option, when it's only benefit is to crash the program.

I thought, if already changed the default, but I'll check.

closed 2020-07-26 by zarath