NanoVNA-Saver/nanovna-saver #286: LIst assignment out of range issue with Segments>1

igor-m, 2020-07-26:
While running 037-pre01 (BW 2kHz, 201p, latest dislord 1.0.2, H4) with SEGMENTS>1 I get following error.
[image]

PA3CCE, 2020-07-26:
I have the same problem, get the errormessage after a multi segments sweep, though the sweep has finished correctly. 
This happened to me as well  while running a calibration,  after the errormessage the calibration window does respond to mouse clicks....I couldn't finish the calibration.
I'm using the latest firmware by DiSlord (Version 1.0.2 beta), it looks like, with this firmware, I have to run new calibrations, because the result is different to a previous measurement..

igor-m, 2020-07-26:
Settings: C:/Users/me/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
2020-07-26 18:32:36,406 - NanoVNASaver.SweepWorker - ERROR - list assignment index out of range
Traceback (most recent call last):
  File "C:\Users\me\NanoVNASaver\nanovna-saver_zarath\nanovna-saver\NanoVNASaver\SweepWorker.py", line 84, in run
    self._run()
  File "C:\Users\me\nanovna-saver_zarath\nanovna-saver\NanoVNASaver\SweepWorker.py", line 136, in _run
    self.updateData(freq, values11, values21, i)
  File "C:\Users\me\nanovna-saver_zarath\nanovna-saver\NanoVNASaver\SweepWorker.py", line 190, in updateData
    self.data11[offset + i] = data11[i]
IndexError: list assignment index out of range

PA3CCE, 2020-07-26:
Yes I have had to replace the CH0 connector of my H4 already, now I'm using portsavers on CH0 an CH1.... I get the same errormessages. I think I'd better delete all my calibration files in NanoVNA-Saver :(

zarath, 2020-07-26:
Gimme Debug.... Shouldn't crash either...

PA3CCE, 2020-07-27:
@Zarath I attached here my debug file.
[debug_3.7_rc02.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4980810/debug_3.7_rc02.txt)

igor-m, 2020-07-27:
I inserted the traceback in my post above..

RandMental, 2020-07-27:
Now that the makers are set to Start, Mid and Stop, one cannot edit them, however moving them on the graphs work.

zarath, 2020-07-27:
recalculated frequencies for sweep.
Bug should be fixed in pre03

closed 2020-07-28 by zarath

igor-m, 2020-07-29:
In 037-pre07 a crash after the initial sweep.
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4994064/debug.txt)

zarath, 2020-07-29:
I see,
somewhere the change to 201 sweep points was missed... I'll take a look.

For your explanation. That "Init data length: 101" fills up the array structure for the incoming data.
So it has to be as large as the expected data.
But the sweep command shows 201 datapoints ..

2020-07-29 11:59:43,230 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-29 11:59:43,231 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(1000000, 2500000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-07-29 11:59:43,231 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (1000000, 2500000000)
2020-07-29 11:59:43,233 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-29 11:59:43,234 - NanoVNASaver.SweepWorker - DEBUG - Sweep segment no 0
2020-07-29 11:59:43,234 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (1000000, 2500000000)
2020-07-29 11:59:43,235 - NanoVNASaver.SweepWorker - INFO - Reading from 1000000 to 2500000000. Averaging 1 values
2020-07-29 11:59:43,235 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
2020-07-29 11:59:43,236 - NanoVNASaver.SweepWorker - DEBUG - Setting sweep range to 1000000 to 2500000000
2020-07-29 11:59:43,236 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(scan 1000000 2500000000 201)

I think I can find the Issue with this debug.txt. Thanks

zarath, 2020-07-29:
@igor-m can you try pre08

igor-m, 2020-07-29:
pre08 works..