NanoVNA-Saver/nanovna-saver #289: Markers - cannot edit prefilled value

igor-m, 2020-07-27:
Cannot edit the Marker field, there is a pre-calculated value (that is nice) which cannot be changed, however.
037-pre01

zarath, 2020-07-27:
should be fixed in pre03

igor-m, 2020-07-27:
When you change the start and stop frequencies the "automatic pre-calculation" does not update the Marker's fields with new values.
You have to place the cursor into the Marker's fields and it shows the new pre-calculated value then..

zarath, 2020-07-27:
You are nitpicking ;)

PA3CCE, 2020-07-27:
Have the same problem running nanovna-saver-0.3.7-**pre04**. so no nitpicking :)
Program starts with start 3.6MHz and stop 30MHz, when I set for different frequency span and run it, it keeps showing 3.6-30MHz in the displays,....Nanovna device itself shows the new frequency span.....I also have problems with setting new marker frequencies.
I had set for a debugfile "debug_3.7_pre04.txt" but it does not show in the repository..... Nothing special in the command window.
Running OK with multi segments.

zarath, 2020-07-28:
No problem. ...
But pre05 will be 100% bug free[tm]

PA3CCE, 2020-07-28:
Are you sure ? (just joking). :)

zarath, 2020-07-28:
Of course, that's why it is called a pre release..

zarath, 2020-07-28:
@PA3CCE can you confirm, that this is fix in >=pre05?

PA3CCE, 2020-07-28:
@ Zarath..Well, first of all I have set for **8** markers, yes I know quite a lot....When I start NanoVNA-Saver all settings are empty.... Then when I connect the device, Start/Stop is 3MHz/102.85 MHz and the first 3 markers are filled 3MHz-52.925MHz-102.85MHz and the remainig 5 markers are all filled with 3MHz !!. Segmments is set to 1.  When I set for 10 Segments nothing changes. ..I can set the first 3 markers, no problem, but the remaining 5 markers,when I try to change them and press enter, the setting jumps back to 3 MHz !! ...Then I start a sweep and when the sweep has finished I can set the last 5 markers to what I want.
I hope this helps.

PA3CCE, 2020-07-28:
Her is my last debug file.

[debug5-rc05.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4990517/debug5-rc05.txt)

zarath, 2020-07-28:
as I told, 100% bug free...

the marker 4-8 setting to start freq was intended...
Perhaps it try to distribute all enabled marker evenly...

zarath, 2020-07-28:
Have fun with v0.3.7-pre07 ;-)

RandMental, 2020-07-29:
We need to think this through before just trying different options. 

 

The idea to set the first 3 makers on start-up to the Start Freq, Centre Freq and  End Freq is to have the makers separated on the Graphs  so one can easily identify them,  select them and move them to the wanted frequencies.

 

Markers 4 and further down, if enabled, should not be given any frequency as it will clutter the graphs and Screen layout with markers that 95% of users will never use.   If they are enable and has been given a frequency – store it in the  application  .ini file so that those users can have an easy start-up and continue from where they left off previously.

 

I would further suggest we add the following keyboard shortcuts to Markers 1 to 3

Marker 1 -   Select and move it on the graphs by using Shift + Left Click and drag

Marker 2 -   Select and move it on the graphs by using Control + Left Click and drag

Marker 3 -   Select and move it on the graphs by using Alt + Left Click and drag

 

 

Regards

Randmental

zarath, 2020-07-29:
> We need to think this through before just trying different options. The idea to set the first 3 makers on start-up to the Start Freq, Centre Freq and End Freq is to have the makers separated on the [...]

That's far in the future ... 

but I think evenly distributed is ok atm. So when starting with the three default markers you get start, center, end.
with two just start, end. with five, start, 1/4, center, 3/4, end

igor-m, 2020-07-29:
https://github.com/NanoVNA-Saver/nanovna-saver/issues/260#issuecomment-659663711

closed 2020-07-29 by zarath