NanoVNA-Saver/nanovna-saver #290: setup.py reports wrong python version ?

JulyJim, 2020-07-28:
This is not a bug 
I am first time user off python and I am trying to trace the code for purpose of learning how NanoVNA-Saver code gets executed.
I am not sure where to post this and if necessary I do not mind if it gets deleted as inappropriate post  here. 
 
I have successfully  cloned this repository to my local repository.
I did try to run " main" and received many errors , basically indication of wrong version of python. 

I did run "setup.py" and it confirmed my suspicion - it detected < 3.7 version . 
But I checked  and I indeed have python3.7 installed. 

Did I clone wrong repository ?

Is running "setup.py" good way to double check the python version ?

I am really sorry to post this here.

zarath, 2020-07-28:
Perhaps you have multiple python versions installed.
Then calling
`python3.7 setup.py`
can help.

JulyJim, 2020-07-28:
Thank you,
Now the question  is how to instruct Eclipse to use correct version of python. 
I am looking at PyDev options now and it should be there somewhere as "interpreter selection ". 
I hope.

JulyJim, 2020-07-28:
BINGO
Works as expected. Dummy picked "default" interpreter during initial install.
Since thnsi was not a bug, can I delete whole exchange ?

closed 2020-07-28 by JulyJim