NanoVNA-Saver/nanovna-saver #291: Conditional errors in NanoVNA-Saver version 0.3.7-rc05

PA3CCE, 2020-07-28:
**Describe the bug**
1) When I set "Bandwidth" in NanoVNA-Saver 0.3.7-rc05 to**10 Hz** I get a freeze ..( See also my warning "Latest firmware by  DiSlord etc.)" .

2) When I set in "SWweep settings"  Average Sweep to **3-0** NanoVNA-Saver runs OK, when I set it to **9-5** I get error message  :

**I'm running the very latest versiion by DiSlord (Version 1.0.3 betaNew UI 192k ADC build time jul26 2620 2020 - 22:32:03)**

see the debug files :

[dbug-rc05.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4988240/dbug-rc05.txt)
[debug2-rc05.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4988243/debug2-rc05.txt)

PA3CCE, 2020-07-28:
Now I restarted NanoVNA-Saver from the same commandwindow and got an errormessage when trying to connect, see debug file.
[debug3-rc05.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4988379/debug3-rc05.txt).

PA3CCE, 2020-07-28:
I managed to do a calibration using Start/Stop 3MHz/2GHz,  20Segments,  201 points,  Bandwidth 33 Hz (don't khow wat was set actually),  averaged sweep 3/0, and measured a BiasTee, yes it took some time.   Attached a screenshot.    :)

[image]

zarath, 2020-07-28:
One glitch with refactoring averages/truncates... and one time again input validation with connecting an uncalibrated device.
Changed in pre06

PA3CCE, 2020-07-28:
OK, thanks, will test.

closed 2020-07-29 by zarath